bluesky/ophyd #456: Non-Signal read attributes

teddyrendahl, 2017-10-19:
Currently, all of the attributes listed under `read_attrs` are queried like so:
```python
obj = getattr(self, attr)
obj.read()
```
I was curious if you had ever considered allowing users to place arbitrary Python properties in the `read_attrs` list, just emitting the value without calling `read` on the returned `getattr`. This allows a pretty quick and easy way for users to implement device side logic. Here is a quick example:

```python
class MyDevice(Device):
       sig1  = C(EpicsSignal, ':SIG1', auto_monitor=True)
       sig2  = C(EpicsSignal, ':SIG2', auto_monitor=True)

       def __init__(self, *args, read_attrs=None, **kwargs):
           if read_attrs is None:
               read_attrs = ['normalized']
           super().__init__(*args, read_attrs=read_attrs, **kwargs)

       @property
       def normalized(self):
           return self.sig1.value / self.sig2.value
``` 

Obviously, you could put `sig1` and `sig2` in the `read_attrs` and normalize the values later, but this seems far simpler. This is especially if true if you wanted to plot `normalized` in a `LivePlot` or similar callback.

The major issue I see is, what do we consider the timestamp of our Python property? Can we just use the timestamp of when we called the property?

 Thoughts?

klauer, 2017-10-19:
You may want to consider using `DerivedSignal` (https://github.com/NSLS-II/ophyd/blob/master/ophyd/signal.py#L244), implementing your own `forward` and `reverse`. It handles all of the other necessary metadata.

tacaswell, 2017-10-19:
And `AttributeSignal` provides a way to mirror out  an attribute.

The read attributes also have to provide a `describe` method to feed the event descriptors.

We did sped a fair amount of time trying to get components closer to properties and did not come up with anything that was better than the status quo

teddyrendahl, 2017-10-19:
Cool. Thanks!

closed 2017-10-19 by teddyrendahl