bluesky/ophyd #467: BUG: No Mover

CJ-Wright, 2017-11-01:
It seems to `Mover` got lost in the move from bluesky to ophyd. It is still referenced in all the examples though. Can we please get a shim for this?

danielballan, 2017-11-01:
No, we intentionally removed `Mover` and `Reader`. These objects were never intended for wide adoption, and I believe @tacaswell has consistently emphasized that they would not be maintained long term and should not be used as a platform. Their detailed behavior differs in important ways from actual hardware. It is much better to test against objects that share as much as possible with real hardware -- with the only difference that the data is generated by Python functions instead of communication with hardware over the network.

Please transition to using `SynSignal`, `SynPeriodicSignal`, and `SynAxis`. You will get much better results when you move between these and real hardware.

Excerpting from [this comment](https://github.com/NSLS-II/bluesky/issues/692#issuecomment-340312145) explaining our thinking:

* `Mover` seems like it supports multi-axis devices (the documentation suggests so) but in fact it doesn't really.
* `Reader` provides a way to hack together a Device-like object with multiple fields, but its API does not resemble the API of ophyd.Device, the one that users would have to use to talk to actual hardware. Therefore, its usefulness as a teaching tool is questionable.

CJ-Wright, 2017-11-01:
You might consider removing it from the [doc strings](https://github.com/NSLS-II/ophyd/blob/master/ophyd/sim.py#L397). I'd be rather interested in those docs since I plan to use them as a model for our simulated hardware.

danielballan, 2017-11-01:
Oh yes, thanks for flagging that. After tagging 1.0.0 today, I plan to pivot to shoring up our documentation. Should take about a week for me to get to that. If you can't wait, take a look at the new docstrings (I believe they are correct) or read the code! 😈

tacaswell, 2017-11-02:
Done in #473 and https://github.com/NSLS-II/bluesky/pull/909

closed 2017-11-02 by tacaswell