bluesky/ophyd #475: Fall back to KILL if STOP fails?

danielballan, 2017-11-02:
See https://github.com/NSLS-II/ophyd/blob/master/ophyd/epics_motor.py#L115

danielballan, 2017-11-02:
I created this issue hurriedly in the middle of a tutorial. Let me now provide a little more context.

The EPICS motor record exposes a STOP field and a KILL field. According to the controls engineers in my tutorial, which of these is more certain to effectively stop the motor can depend on the kind of motor it is. Sometimes a powered abort is safer; other times just killing it is more sure to work.

It might be nice to make this configurable through an argument to `__init__`, such as `stop_mode='kill'`. Or, alternatively, to generically STOP, try to detect whether that seems to be working, and then possibly KILL if needed.

Of course, it is currently possible to customize this by subclassing `EpicsMotor` and overriding the `stop()` method. Perhaps the KILL case is rare enough that subclassing is a reasonable approach and no new argument is needed. I'll be the first the say I have _no idea what I'm talking about here_ but I wanted to kick off a conversation.

dchabot, 2017-11-02:
Small, but perhaps significant correction: the Motor Record does _not_ have a KILL field. There is a supplementary support record included for the Delta Tau controllers that _does_ have a kill command ($(SYS)$(DEV):Cmd:Kill-Cmd).

IIRC, for those controllers, the STOP field issues a command to "decelerate to a controlled stop", and the kill record means "kill output signal to motor" (i.e. coast to a stop).

klauer, 2017-11-02:
The closest analog is the `.CNEN` field, which could open-loop the device. It's built into the motor record, but support for it is spotty on a controller-by-controller basis. 

It again would have to be optional, as open-looping motors may be problematic (depending on their mechanical design).