bluesky/ophyd #477: FakeEpicsPV Callbacks

ZLLentz, 2017-11-02:
I am currently in the process of migrating SLAC's devices for compatibility with latest (we had previously frozen the version for a bit while we worked on other things). I'm having trouble migrating our unit tests, which previously used a modified version of `FakeEpicsPV` but now include the class from ophyd's `test/conftest.py` due to all the changes with the `control_layer`.

My issue is that the class's callbacks are stored in `WeakValueDictionary`:

https://github.com/NSLS-II/ophyd/blob/master/ophyd/tests/conftest.py#L51-L54

Keeping this as a `WeakValueDictionary`, all of my callbacks are being instantly garbage collected, so I can't write unit tests for things like returning a working `Status` object after a `set` call if they depend on an EPICS subscription.

My questions are:
1. What is the purpose of using a `WeakValueDictionary` here?
2. What are the negative consequences of changing this to a normal `dict` object? (my tests and the ophyd tests pass with this change)
3. If there are consequences, do I need to handle this by keeping a reference to every callback in all of my device implementations?

klauer, 2017-11-02:
The purpose was to allow PV instances to be garbage collected. Holding strong references did not allow for garbage collection, at least with the version of Python available at the time of writing. 

Perhaps the implementation is "incorrect" as it does not mirror pyepics and its poor handling of garbage collection.

ZLLentz, 2017-11-02:
Python refuses to garbage collect objects that contain strong references to other objects? Surely this means that very few objects would ever be garbage collected, but maybe I am misunderstanding something. Am I making my objects un-garbage-collectable every time I include a dictionary attribute?

klauer, 2017-11-02:
If I recall correctly, it's having a strong reference to a bound instance method in the dictionary that caused it. I'm not sure if that even still applies in the current codebase.

tacaswell, 2017-11-03:
If there is a hard-ref to an object it will not be garbage collected (modulo garbage cycles which it can detect and take care of).   This all gets complicated by having `__del__` (see https://stackoverflow.com/questions/41851098/how-does-del-interfere-with-garbage-collection ), but should be ok in python 3.

I think the problem here is that bound methods are ephemeral (see https://docs.python.org/3.6/library/weakref.html#weakref.WeakMethod ), we probably want to manage this our selves to use either a weakref or weakmethod as required.

I find it very surprising that we do not exercise the status objects with the fake pv class!

tacaswell, 2017-11-03:
I'm working on a fix for this.

ZLLentz, 2017-11-03:
One more related issue I found, an occasional deadlock in the callback loop. If a `FakeEpicsPV` callback needs the `EpicsSignal` lock, but another thread calls `EpicsSignal.put`, we get two threads that need to acquire each other's lock, since the callbacks are run within the `FakeEpicsPV` lock context.

https://github.com/NSLS-II/ophyd/blob/d63d22789dbbf6fad36312008df982e267b190c7/ophyd/signal.py#L881-L890

https://github.com/NSLS-II/ophyd/blob/d63d22789dbbf6fad36312008df982e267b190c7/ophyd/tests/conftest.py#L209-L214

https://github.com/NSLS-II/ophyd/blob/d63d22789dbbf6fad36312008df982e267b190c7/ophyd/tests/conftest.py#L99-L111

teddyrendahl, 2017-11-05:
Reopening as to not lose track of the race condition note above

PR #478 (2017-11-05): Fix fake PV
I sneaked in some style and warning changes as well.

closes #477
commit f013fd45 (2017-11-05): TST: fix FakeEpicsPV callback handling

closes #477