bluesky/ophyd #486: EpicsSignal based on multiple signals

jrmlhermitte, 2017-11-17:
Say you have a signal made of multiple signals:
```python
sig1 = EpicsSignalRO(pvname1, name=name1)
sig2 = EpicsSignalRO(pvname1, name=name1)

```

how to create a new signal such that the result is `sig1.value+sig2.value - bg` where `bg` is some precomputed parameter?

cowanml, 2017-11-17:
On Fri, 17 Nov 2017, Julien Lhermitte wrote:
> 
> Say you have a signal made of multiple signals:
> 
> sig1 = EpicsSignalRO(pvname1, name=name1)
> sig2 = EpicsSignalRO(pvname1, name=name1)
> 
> how to create a new signal such that the result is sig1.value+sig2.value - bg where bg is
> some precomputed parameter?

If that new signal isn't a one-off one time use thing, you're best off
pushing stuff like that down into epics so it's available like a
regular pv to any pv client.

-matt

danielballan, 2017-11-17:
There are a couple considerations here:

1. Is the computation established and stable, or is it part of an experimental workflow that still requires some improvisation? Doing this in the Python layer is much better interactive exploration.
2. Is this something that is straightforward to do in C, or would numpy/scipy/scikit-image/scikit-learn help?
3. Is this something that other users at the facility would want to check or monitor? Consider EPICS!

To your question, ophyd provides DerivedSignal and AttributeSignal. I can never remember which is which! Better documentation is coming. For now, grep the test suite.

cowanml, 2017-11-17:
On Fri, 17 Nov 2017, Dan Allan wrote:
...
>  1. Is the computation established and stable, or is it part of an experimental workflow
>     that still requires some improvisation? Doing this in the Python layer is much better
>     interactive exploration.

sure, although I worry about temporary solutions becoming standards.
You can do similar stuff in css and I tell people the same thing.

>  2. Is this something that is straightforward to do in C, or would
>     numpy/scipy/scikit-image/scikit-learn help?

If it's the simple example mentioned, pv1+pv2-pv3, no c required,
simple epics calc record.

>  3. Is this something that other users at the facility would want to check or monitor?
>     Consider EPICS!

Or that you might want in another process or css or whatev, doesn't
have to be for other people to bother with epics.

I'm not trying to preach epics, just reminding it's there, has some
nifty capabilities/benefits, and isn't always painful.

-matt

danielballan, 2017-11-17:
Thanks for that useful perspective, @cowanml.

jrmlhermitte, 2017-11-17:
yes thanks @cowanml this reminder is appreciated, please do keep giving feedback! :-)

jrmlhermitte, 2017-11-17:
PS: I will contact the controls group on such an instance and close this issue once resolved.
In the meantime, a pseudo-example (with a *big* disclaimer to contact the controls group) could be something to PR, I'll work on that later.

tacaswell, 2019-10-17:
Closing as stale.

If we really need this at the python level this is a pseudo-positioner, but it should be done in the IOC when ever possible.

closed 2019-10-17 by tacaswell