bluesky/ophyd #488: Cleave Sim from EPICs

CJ-Wright, 2017-11-28:
Would it be possible to cleave the simulation module from epics?
When importing `ophyd.sim` I get the following error:
```python
Traceback (most recent call last):
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/test_tmp/run_test.py", line 11, in <module>
    import ophyd.sim
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/sim.py", line 860, in <module>
    globals().update(hw().__dict__)
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/sim.py", line 774, in hw
    motor = SynAxis(name='motor')
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/sim.py", line 269, in __init__
    super().__init__(name=name, parent=parent)
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/device.py", line 733, in __init__
    [getattr(self, attr) for attr, cpt in self._sig_attrs.items()
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/device.py", line 734, in <listcomp>
    if not cpt.lazy]
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/device.py", line 176, in __get__
    instance._signals[self.attr] = self.create_component(instance)
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/device.py", line 131, in create_component
    cpt_inst = self.cls(parent=instance, **kwargs)
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/signal.py", line 56, in __init__
    from . import control_layer as cl
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/control_layer.py", line 13, in <module>
    from ._pyepics_shim import setup, caput, caget, get_pv, pv_form, thread_class
  File "/home/christopher/mc/conda-bld/ophyd_1511857338519/_t_env/lib/python3.6/site-packages/ophyd/_pyepics_shim.py", line 3, in <module>
    import epics
```

danielballan, 2017-11-28:
It's our intention that `ophyd.sim` be importable without pyepics or libca installed. This is a bug.

tacaswell, 2017-11-29:
You have to have _a_ control layer configured and it defaults to pyepics.  There is a separate check if this works with or without libca.   This is the current expected behavior, just install pyepics.

danielballan, 2017-11-29:
I could be misremembering our design discussion. I think @tacaswell is right. It’s possible run ophyd without installing pyepics, but the best way forward in this case is to just install pyepics.

stuartcampbell, 2017-11-29:
My initial reaction is that long term, a true `ophyd.sim` should be able to run completely standalone without requiring a backend to be present.

tacaswell, 2017-12-01:
We can provide a 'dummy control' layer that provides the thread class and a `get_pv` that raises when used.

danielballan, 2018-03-08:
See #520. attn @CJ-Wright

danielballan, 2018-05-02:
Closed by #520. Do let us know if this solves your problem @CJ-Wright because I'm going to run into this problem too, during tutorials.

closed 2018-05-02 by tacaswell