bluesky/ophyd #490: Get access to dial value of motor from bluesky

gmysage, 2017-12-04:
Hi, I have a question. For motors defined by EpicsMotor, how can I get its encoder value (if it has encoder)? From Epics, these value is associated with extension of .DRBV and .RRBV.  

Specifically, .DRBV is the DIAL value from the encoder that reads the absolute value of the motor position, which is independent of the user_set. The unit of DRBV is the same as motor position, e.g. in mm.

And .RRBV is the absolute counts of the encoder reading. So the value of RRBV and DRBV is one-to-one mapped by some internal conversion.

For example, if I have a motor:
mot = EpicsMotor('prefix{Ax:X}Mtr', name='mot'), the encoder value is defined as mot.DRBV and mot.RRBV.

Please kindly give some suggestions.

Thanks,
Ming

danielballan, 2017-12-04:
As I discussed with @gmysage over email, we don't currently expose these specific fields. We do [expose fields related to the dial value](https://github.com/NSLS-II/ophyd/blob/master/ophyd/epics_motor.py#L50) but I'm not sure how they related to `.DRBV`. I don't think we expose `.RRBV`. I'd like to understand better why we would want to.

Our design philosophy has been to expose any fields needed for data acquisition and a few fields that happen to be useful for debugging, but to stop short of a comprehensive set of _everything_ that EPICS can tell us about motor.

Comments from our more controls-oriented contributors would be welcome.

tacaswell, 2017-12-05:
The quickest thing is to just define a local motor sub-class as

```python
class MyMaybeDangerousMotor(EpicsMotor):
    dial_readback =  Cpt(EpicsSignalRO, '.DRBV')
    absolute_encoder_counts = Cpt(EpicsSignalRO, '.RRBV')
```

which will expose those fields to ophyd/bluesky however I suspect that upstream we will eventually provide more complete motor record classes which may pick different names for those fields so you will have to adapt to that eventually.