bluesky/ophyd #502: Add hints to Device

jrmlhermitte, 2018-01-09:
@danielballan @mrakitin 

> Long-term fix: move `hints` into the `Device` class so that this doesn’t keep happening, as we discussed yesterday at ISR, with a property, a setter, and a class attribute `_default_hints`. Short-term fix: copy the pattern from ISR.

Current fix at ISR is [here](https://github.com/NSLS-II-ISR/isr_profiles/blob/master/profile_collection/startup/20-area-detectors.py#L105). This should be eventually moved to `Device` class

teddyrendahl, 2018-01-09:
Absolutely! Linking #487 which seems similar. Currently `hints` require a lot of code duplication.

mrakitin, 2018-05-03:
Fixing the link: https://github.com/NSLS-II-ISR/profile_collection/blob/master/startup/20-area-detectors.py#L105

PR #541 (2018-05-14): Component Kind
Closes #540 Closes #539 Closes #502 Closes #487 

It is possible we make poor life choices, but this seems to work and has a number of additional nice properties

 - the ground truth of if a component is included in the `read` or `read_configuration` (or both!) walks of the Device tree is now held in the components
 - it is now clear how to configure this at class definition time
 - the old API of _setting_ `read_attrs` and `configuration_attrs` still works
 - changing the state of a parent via `dev.child.kind = Kind.OMITTED` won't affect grand chlidren.
 - leaf-nodes can be mad [... cut]
closed 2018-05-14 by danielballan