bluesky/ophyd #518: Ophyd Docs Missing Info About setup_ophyd

ZLLentz, 2018-03-01:
@klauer showed us the `MonitorDispatcher`, which seems to make our `ophyd` sessions with the `pyepics` control layer much more reliable, and explains a lot of race conditions we were scratching our heads over the last few months.

With a docstring like https://github.com/NSLS-II/ophyd/blob/master/ophyd/_pyepics_shim.py#L126-L130 that indicates it should always be called, I'm surprised that `ophyd.setup_ophyd` is not mentioned anywhere in the docs.

This setup routine should either be automatically run or included in the docs.

danielballan, 2018-03-08:
Belatedly -- @ZLLentz Sorry to hear that you guys lost a bunch of time on this. You are right: it should be prominent in the documentation. I'm drafting a tutorial, and this is in it.

tacaswell, 2018-03-09:
#520 also deprecates the need to do this, might be better to push on that?

danielballan, 2018-03-09:
Oh, has your thinking changed on setup-on-init? I thought we avoided that intentionally. From a user's point of view it seems nice, but there's a risk of getting into a state where ophyd crashes Python on startup.

tacaswell, 2018-03-09:
I think it is on ophyd to make sure setup is safe!  I think the historical issues were from the imports from libca not being sufficiently wrapped in try-except.

klauer, 2018-03-09:
Right - the `setup()`-on-import functionality was removed at your request.

danielballan, 2018-03-09:
> I think the historical issues were from the imports from libca not being sufficiently wrapped in try-except.

I see. Seems like that fine with @klauer and it's fine with me, just wanted to understand what changed.

closed 2018-05-02 by danielballan

tacaswell, 2018-05-08:
@klauer In an case I have changed my mind and decided you were correct to begin with.

commit 2fce2ae8 (2018-05-02): API move control layer selection logic to __init__

Automatically call setup when selecting the control layer

closes #518