bluesky/ophyd #521: Can all ophyd objects have a simulation mode?

CJ-Wright, 2018-03-08:
Would it be possible that all ophyd objects have a simulation mode (maybe triggered by an env var?).
The rational is that we could then copy entire beamlines' `ipython_ophyd` objects over to a simulation system (CI?) and run tests.

tacaswell, 2018-03-08:
The biggest problem is how to simulate the correct interaction between the PVs (for example, if you set the acquire bit to high on a area detector you would like it to flip back to low after the delay on acquire time).  How high of a fidelity simulation you want is an open question. 

Right now the ophyd objects to not encode this information in any easily programitically accessible way (some of it is in the source of callback functions, ex what flips the status object to done when the acquire bit goes low and some of it is in the source of the IOC which is (from ophyds point of view) a black box).

If you just want to be able to import a profile you can use https://github.com/NSLS-II/caproto/blob/master/scripts/spoof-beamline.

You could also write a control layer that is smarter than dummy, but provides fully synthetic PVs (a-la https://github.com/NSLS-II/ophyd/pull/520 ), but that does not get around the problem of not knowing the linkages between the PVs.

There is on-going work to make standing up complex interconnected servers easier (https://github.com/NSLS-II/caproto/pull/157 ) but that is still a bit out.

cowanml, 2018-03-08:
probably easier to use lower level simulators in many cases?
https://www.google.com/search?q=epics+area+detector+simulator+ioc

On Thu, 8 Mar 2018, Thomas A Caswell wrote:
...
> There is on-going work to make standing up complex interconnected servers easier
> (NSLS-II/caproto#157 ) but that is still a bit out.

Somehow this bit reminds me of "Starter devices" in Tango.
http://www.esrf.eu/computing/cs/tango/tango_doc/ds_doc/tango-cs/System/starter/Version.html

-matt

teddyrendahl, 2018-03-08:
At SLAC we are interested in this as well. The fact that the `ophyd.sim` objects are actual Ophyd objects has gone along way towards this goal. Still it would be nice if there was a programatic way to rip out `EpicsSignal` objects and replace them with `SynSignal`. Our goal wouldn't be the simulation of individual devices but the relationship between motors and detectors to help test high level automation.

klauer, 2018-11-16:
This is done rather easily with `ophyd.sim.make_fake_device` now and is also possible through simple caproto-based IOCs. Examples are all over the test suite, especially in #646.

closed 2018-11-16 by klauer