bluesky/ophyd #527: QuadEM currentAll

danielballan, 2018-04-08:
From an email from Lin Yang:

> Could you add currentAll as one of the attributes of the quadem class, when you update Bluesky/Ophyd? The PV does exist.

We should look at the QuadEM variant used at ESM and BMM to make sure this hasn't already been done there.

jrmlhermitte, 2018-05-10:
ESM [here](https://github.com/NSLS-II-ESM/profile_collection/blob/master/startup/30-detectors.py)
BMM [here](https://github.com/NSLS-II-BMM/profile_collection/blob/master/startup/40-quadem.py)

doesn't look either has done this? I think PR in #537 would do it, but it might break if this PV doesn't exist

We could remove the attr after the fact if that's the case. So what's your opinion on this?

jrmlhermitte, 2018-05-10:
closed by #537

closed 2018-05-10 by jrmlhermitte