bluesky/ophyd #535: path semantics to write_path_template in TIFFPlugin?

jrmlhermitte, 2018-05-07:
There is an issue with the code [here](https://github.com/NSLS-II/ophyd/blob/master/ophyd/areadetector/filestore_mixins.py#L49).

When dealing with windows machines, this should add a trailing backslash "\" and  not forward slash "/".
This is because for some area detector plugins on windows machines, a backslash is added.

This will cause the setting string to not match the string area detector actually sets to causing a hanging `set_and_wait`.

I suggest adding a kwarg `path_semantics="posix"` to this function

stuartcampbell, 2018-05-07:
Is this just a problem with the TIFFPlugin ?  Are newer versions the same ?

Should we not just try and fix this upstream ?  Windoze is happy with either '/' or '\\' as long as it's consistent.

jrmlhermitte, 2018-05-07:
Thanks for the feedback!
I emailed Li Ji to ask if we can avoid it. Here's what's happening:
```py3
xf28id2@xf28id2-ws3:~$ caput -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath "foo"
Old : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foo
New : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foo
xf28id2@xf28id2-ws3:~$ caget -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV 
XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV foo\\
```
I also asked him for the versions.

tacaswell, 2018-05-08:
`set_and_wait` should not be aware of path details.

jrmlhermitte, 2018-05-08:
I think I may not have explained correctly. This is independent of `set_and_wait`. The problem is two layered (in this order):
1. `write_path_template` property is adding a trailing slash "/" if not present
2. This particular area detector tiff plugin we're using happens to add an extra trailing backslash "\" if not present.

To fix number 2, all we need to do is make sure our file path ends in "\". However, because of 1, this becomes impossible. We end up in a situation where it is impossible for the set PV and the readback PV to ever match. Examples:
- 'foo' -> 'foo\'
- 'foo\' -> 'foo\/\'
etc

I think if `write_path_template` includes logic that knows about paths (i.e. adds a trailing slash), then it should maybe know the difference between windows and linux. We can probably find a way to fix this ioc this time around but there are no guarantees we won't encounter this problem again.

I'm waiting for the response from Ji Li just in case.

jrmlhermitte, 2018-05-08:
I remember how we solved it. We stopped reading from the RBV. (So changed from Epics with RBV to regular EpicsSignal). This should more or less explain the issue:
```
# set the filepath PV
xf28id2@xf28id2-ws3:~$ caput -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath "foof/fdasf/fdas/fdsa/"
Old : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foof/fdasf/fdas/fdsa/
New : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foof/fdasf/fdas/fdsa/
# read from the RBV
xf28id2@xf28id2-ws3:~$ caget -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV
XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV foof/fdasf/fdas/fdsa\\
```

If we don't use the RBV, then we can refer to paths with "/".
Ideally, it would be nice to use the RBV. I have emailed Mark Rivers to ask him.

tacaswell, 2018-05-08:
The reason we read the RBV is to make sure that we and the IOC agree on what the value is.  We discovered many race conditions where the set PV would be updated, but the internal state would not be.  Most of these are things you can not loose if you are clicking in CSS, but bluesky loses them all the time....

jrmlhermitte, 2018-05-09:
Thanks for the feedback. I'm 100% on using the RBV. The workaround is not to use the RBV, which maybe XPD did (have to go can check later). Here is a conversation between me and MarkRivers:

> Hi Julien,
> 
> I was mistaken.  I seem to recall that the code behaved as I described sometime in the past, but now it always terminates the path with "\" on Windows.
> 
> Mark
> 
> 
> -----Original Message-----
>XXXXXX
> Subject: RE: PE 4k
> 
> Mark,
>    Unfortunately, I'm not able to reproduce what you mentioned. For the set PV there is no issue. However, for the RBV there is.
> Here is the issue:
> # set the filepath PV
> xf28id2@xf28id2-ws3:~$ caput -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath "foof/fdasf/fdas/fdsa/"
> Old : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foof/fdasf/fdas/fdsa/ New : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foof/fdasf/fdas/fdsa/ # read from the RBV xf28id2@xf28id2-ws3:~$ caget -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV
> XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV foof/fdasf/fdas/fdsa\\
> 
> 
> Perhaps I am doing something wrong?
> thank you for your time, I much appreciate it.
> (I am not very experience with AD)

jrmlhermitte, 2018-05-09:
Figured out the problem. There is a difference in the plugin versions. I couldn't get the version info, but I can show the inconsistency with these commands:

The old way with pe1 worked as such:
```bash
(py3) lhermitte@xf28id2-ws3:~$ caput -S XF:28IDC-ES:1{Det:PE1}TIFF1:FilePath "foof/fdasf/fdas/fdsa/"
Old : XF:28IDC-ES:1{Det:PE1}TIFF1:FilePath H:/pe1_data/2018/05/09/
New : XF:28IDC-ES:1{Det:PE1}TIFF1:FilePath foof/fdasf/fdas/fdsa/
(py3) lhermitte@xf28id2-ws3:~$ caget -S XF:28IDC-ES:1{Det:PE1}TIFF1:FilePath_RBV
XF:28IDC-ES:1{Det:PE1}TIFF1:FilePath_RBV foof/fdasf/fdas/fdsa/
```

But the new PV's readback doesn't appear as it should:
```bash
(py3) lhermitte@xf28id2-ws3:~$ caput -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath "foof/fdasf/fdas/fdsa/"
Old : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath E:/pe2_img/2018/05/09/
New : XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath foof/fdasf/fdas/fdsa/
(py3) lhermitte@xf28id2-ws3:~$ caget -S XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV
XF:28IDC-ES:1{Det:PE2}TIFF1:FilePath_RBV foof/fdasf/fdas/fdsa\\
```

A monkey patch is [here](https://github.com/NSLS-II-XPD/ipython_ophyd/commit/5e53cb8394fd17a339bba6cf0bf50c6807307070#diff-04bbe4324b4170dd2e65058217929393R17) and I suggest pushing this upstream.

mrakitin, 2018-05-25:
I've found the suggested fix is not working for me at XPD:

https://github.com/NSLS-II-XPD/ipython_ophyd/blob/5e53cb8394fd17a339bba6cf0bf50c6807307070/profile_collection/startup/80-areadetector.py#L25-L29

The problem here is: `TypeError: 'str' object does not support item assignment`.

I think the more proper fix would be:
```py
newpath = newpath[:-1] + '\\'
```

closed 2022-01-13 by jrmlhermitte