bluesky/ophyd #540: Component contains grouping information

teddyrendahl, 2018-05-11:
Spinning off discussion from  https://github.com/NSLS-II/ophyd/pull/539.  Curious what people thought of putting some of the logic that is in `_default_read_attrs` e.t.c into the Component itself. The result might look like this

```python
class Motor(Device):
      readback = Cpt(EpicsSignalRO, '.RBV', cpt_type='hint')
      setpoint = Cpt(EpicsSignal, '.VAL', cpt_type='read')
      offset = Cpt(EpicsSignal, '.OFF', cpt_type='configuration')
```
 Might want to make `cpt_type` accept an `enum` not a string to limit input, but this might save some if we start wanting to add more ways to categorize components. This would mean the `Device` metaclass could assemble the default attributes itself.  Obviously API is not permanent. Just wanted to write this out before I forgot.

tacaswell, 2018-05-11:
I like this.

Also see 'prompt' and 'promptgroup' in dbd files [1] 

[1] https://epics.anl.gov/base/R3-16/1-docs/AppDevGuide/DatabaseDefinition.html#x7-2090006.2

danielballan, 2018-05-11:
This is great. I think it's probably much easier for user to deal with `motor.readback.cpt_type = 'read'` that manipulate the elements in lists like `motor.read_attrs`. Instead of memorizing `*_attrs`, as well as all the list-mutation methods, they only have to know `cpt_type =`. It also doubles down on, "Local configuration is better than global configuration." The aggregate lists of which components are in which group can still be generated and displayed, as `motor.summary()` does.

My initial reaction: I think we should strongly consider slowly deprecating `read_attrs` and `configuration_attrs` in favor of this approach.

One wrinkle I'm not sure what to do with: a component that is of type 'hint' is also always in type 'read'. Is there a better way to express that?

klauer, 2018-05-11:
I like the suggestion quite a bit. I think `cpt_type` should determine the `default_x_attrs`, but thought should be given to “how can I modify that behavior in a subclass in a sensible/easy way?”

Additionally, multiple component types should be specifiable - and I’d recommend using a `enum.Flag` for that.

danielballan, 2018-05-11:
Notes on a the video conversation:

* NSLS-II intends to try deploying this on Monday and see how it goes.
* This backs us out of a couple problems are we are seeing at the beamlines:
    * The `read_attrs`settings are scattered far apart from device definitions, leaving scientists/users not sure where to look to change the behavior of a given component.
    * The `hints` are not synced carefully with `read_attrs`, leading to inconsistencies that can break consumers.
    * Managing large lists of names is cumbersome.
* Component passes `kind` through to its argument, just as the same as it does with `timeout` or `settle_time`. In a subclass, you cannot modify the `kind` of a Component defined by a superclass. But, this is already true of `timeout` and `settle_time`. Work-arounds to this limitation all seem messier than either setting it in `__init__` or overwriting the class member by re-defining it.
* The names `read_attrs` and `configuration_attrs` always had a weird asymmetry with `read()` and `read_configuration()`. The name `read_attrs` in particular is not very intutive. We are going with something like kind={`OMITTED`, `NORMAL`, `CONFIG`, or `HINTED`}. The brevity of these was a priority. Via the magic of `IntFlag`, something that is `HINTED` is always also `NORMAL` (so that it is read). Things can be `NORMAL`, `CONFIG`, both, or neither (`OMITTED`).
* Back-compatibility for `read_attrs` and `configuration_attrs` is provided through evil "list" objects that supports the list API (modulo ignoring user-specific order and any repeated items) but, under the hood, just reflect and as needed update the `.kind` attributes of a Device's components. These objects maintain no state on their own. If we need _perfect_ back-compatibility we can add it, but the consensus is that this sufficient for our purposes and the right amount of complexity.
* The current system makes it easy to give a short whitelist of read_attrs a.k.a. Components of a `NORMAL` kind. The new system should support that too. Now it does, via a commit added after the phone call.

tacaswell, 2018-05-13:
One thing that you can do with the current system that you can mix direct child and grand (, grand ...)-child component names in `read_attrs` and it does the right thing (because `getattr(grand_child).read()` does what you expect).  This 'generation skipping' is convenient, but can lead to confusingly set up devices where a child is not in it's parents `read_attrs` and it's `read_attrs` are badly configured, but via this generation skipping one of it's children is in the parents `read_attrs` and the top-level device mostly does the right thing (however it is my suspicion that we are not reading all of the configuration we should be).

We can support grand-children via the `read_attrs.setter` in a (hopefully) non-controversial way be detecting them add then re-cursing : 
https://github.com/danielballan/ophyd/blob/179f69ac671b277c0dc1f07c2d9a5994a5b75647/ophyd/device.py#L819-L827
this has the effect of 'correctly' setting up the `read_attrs` list of the children.  The new system also ensures that sub-devices are in their parents `configuration_attrs` if they are in the `read_attrs`.  This is a small API change, but arguably a bug-fix.  

<details>
The only thing I think this would break is someone being much too clever and using

```python
dev.read_attrs = ['grand.child']
dev.grand.read_attrs = ['some', 'other', 'children']
```
and expecting `dev.read()` and `dev.grand.read()` to have completely dis-joint keys.  I don't really see a use-case for that and the new behavior of forcing `dev.grand.read()` being a strict sub-set of `dev.read()` makes a lot more sense.
</details>

so far so good, but a maybe bigger hitch is that we can not support arbitrarily mixed-generation `read_attrs` (that is `dev.read_attrs = ['child1', 'child2.a']`).  We have 3 choices we can support with the new 'truth lives on the components' scheme (but they all change behavior)

 1. only include direct children.  This mean if the user does `dev.read_attrs = ['grand.child']`, then the accessed value would be `dev.read_attrs == ['grand']
 2. recuse down to the leaves and _only_ include the end signals.  In this case the length of `read_attrs` and the length of `dev.read()` are the same and `[getattr(dev, n).name for n in dev.read_attrs]` gets you back the read keys.  However this means doing somethnig like `dev.read_attrs = ['grand']` would result ind `dev.read_attrs == ['grand.child1', 'grand.child2']`.  This has the weird side effect that if no child devices have any configuration to offer, then `dev.configuration_attrs` is empty which is confusing.
 3. include _all_ generations of names, basically combine 1 and 2.  

I am between 2 and 3

No matter which of the above we pick, we can (should) make it so that `'child' in dev.read_attrs` gives the right answer if we will walk across `dev.child` as part of `dev.read()` and make it so that `dev.read_attrs = [...]` behaves correctly.

danielballan, 2018-05-13:
Having played with many configurations of nested devices in the new suite of tests, I strongly advocate for (3) being the least confusing option.

klauer, 2018-05-13:
Agreed on option (3).

PR #541 (2018-05-14): Component Kind
Closes #540 Closes #539 Closes #502 Closes #487 

It is possible we make poor life choices, but this seems to work and has a number of additional nice properties

 - the ground truth of if a component is included in the `read` or `read_configuration` (or both!) walks of the Device tree is now held in the components
 - it is now clear how to configure this at class definition time
 - the old API of _setting_ `read_attrs` and `configuration_attrs` still works
 - changing the state of a parent via `dev.child.kind = Kind.OMITTED` won't affect grand chlidren.
 - leaf-nodes can be mad [... cut]
closed 2018-05-14 by danielballan