bluesky/ophyd #542: Consider allowing multiple kinds specified as string.

danielballan, 2018-05-14:
Lifting @jrmlhermitte's suggestion from the line comments in #541:

> if provided in string format, I'd be more tempted to allow for a list of kinds:
`kind='normal,hinted'`

tacaswell, 2018-05-14:
If we are going to go down this path, then it would make more sense to us `|` as the separator as that is how we will combine them under the hood.  But if we do that then we might want to also include the `&` operator as well, and then we are on our way to writing a mini-language.

danielballan, 2018-09-11:
Given @tacaswell's point above I think this is not worth doing. Once people need a union/intersection whatever, it's time to just use Python instead of the string shortcut.

If anyone wants to continue debate, feel free to reopen the issue.

closed 2018-09-11 by danielballan