bluesky/ophyd #566: motor labels in Devices

prjemian, 2018-06-01:
Was hoping that the new `labels` feature would work in `Devices`, too.  Is my setup wrong?  I'm looking for additional motors ``sample_stage.{rotary,x,y,pitch,roll}``

## setup

```
from ophyd import MotorBundle

class SampleStage(MotorBundle):
    rotary = Component(EpicsMotor, "2bmb:m100", labels=("theta", "sample"))
    x = Component(EpicsMotor, "2bmb:m63", labels=("sample",))
    y = Component(EpicsMotor, "2bmb:m57", labels=("sample",))
    pitch = Component(EpicsMotor, "2bmb:m54", labels=("sample",))
    roll = Component(EpicsMotor, "2bmb:m53", labels=("sample",))
    changer_status = Component(EpicsSignalRO, "2bmb:Begin_Scan")
    position_status = Component(EpicsSignalRO, "2bmb:Sample_Stage_Position")

sample_stage = SampleStage(name="sample_stage")

camera_rail = EpicsMotor("2bmb:m31", name="camera_rail", labels=("camera",))
top_z_0 = EpicsMotor("2bmb:m76", name="top_z_0", labels=("top",))
top_x_90 = EpicsMotor("2bmb:m77", name="top_x_90", labels=("top",))
focus = EpicsMotor("2bmb:m78", name="focus", labels=("camera",))
```

## response

```
In [2]: wa
camera
        Positioner                     Value       Low Limit   High Limit  Offset     
        camera_rail                    200.0       20.0        1075.0      185.0      
        focus                          116.11825   -637.43175  322.56825   122.56825  
top
        Positioner                     Value       Low Limit   High Limit  Offset     
        top_x_90                       -11681.40075 -1000011625.3975 9999988374.6025 -12178.5505
        top_z_0                        17607.90125 -99982133.61125 100017866.38875 17866.38875
```

tacaswell, 2018-06-01:
I suspect we till have some bugs in how the recursion works....

danielballan, 2018-06-01:
That looks like the output of bluesky before we fixed the bug in recursion. What's the `bluesky.__version__`, @prjemian?

prjemian, 2018-06-02:
BlueSky version: 1.3.1

```
[user2bmb@lyra,42,~]$ use_bluesky.sh 2bmb
BlueSky executable path: /home/beams/USER2BMB/Apps/BlueSky/bin
Python 3.6.5 :: Anaconda, Inc.
conda 4.5.4
pip 10.0.1 from /home/beams/USER2BMB/Apps/BlueSky/lib/python3.6/site-packages/pip (python 3.6)
starting ipython with profile: 2bmb
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: 2bmb
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/00-0-checks.py
BlueSky version: 1.3.1
Ophyd version: 1.2.0a2
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/00-startup.py
Loading metadata history from /home/beams/USER2BMB/.config/bluesky/bluesky_history.db
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/01-databroker.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/02-pyepics.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/10-imports.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/11-motors.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/15-custom-devices.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/20-signals.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/25-PG3-grasshopper.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/30-busy_fly_scan.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/60-metadata.py
using profile:  2bmb
Metadata dictionary:
RE.md['APS_BlueSky_tools_VERSION'] = 0.0.19+5.g943bbe1
RE.md['BLUESKY_VERSION'] = 1.3.1
RE.md['EPICS_APP'] = /net/s2dserv/xorApps/epics/synApps_5_6/ioc/2bmb
RE.md['EPICS_AR_PORT'] = 7002
RE.md['EPICS_CA_MAX_ARRAY_BYTES'] = 100000000
RE.md['EPICS_EXTENSIONS'] = /APSshare/epics/extensions
RE.md['EPICS_HOST_ARCH'] = linux-x86_64
RE.md['OPHYD_VERSION'] = 1.2.0a2
RE.md['SESSION_STARTED'] = 2018-06-01 18:59:41.655305
RE.md['USER2BMB_ROOT_DIR'] = /local/user2bmb
RE.md['beamline_id'] = 2-BM tomography
RE.md['ipython_profile'] = 2bmb
RE.md['login_id'] = user2bmb@lyra.xray.aps.anl.gov
RE.md['pid'] = 25266
RE.md['profile'] = 2bmb
RE.md['proposal_id'] = 
RE.md['scan_id'] = 73
```

prjemian, 2018-06-02:
got these versions via `conda install ophyd bluesky databroker -c lightsource2-tag`

danielballan, 2018-06-02:
Oh, good. We released v1.3.2 to fix this specific problem. It is available on https://anaconda.org/lightsource2-tag/bluesky . Can you verify that the fix solves your problem?

danielballan, 2018-06-02:
Release notes: https://nsls-ii.github.io/bluesky/api_changes.html#v1-3-2-2018-05-24

prjemian, 2018-06-04:
Yuppers, that fixes it.  Is this issue a duplicate of another?

closed 2018-06-04 by prjemian

danielballan, 2018-06-04:
I don't think it ever had an issue; we went straight for a PR to fix: https://github.com/NSLS-II/bluesky/pull/1037