bluesky/ophyd #571: Signal Introspection

teddyrendahl, 2018-06-07:
## Description
It would be nice if `ophyd.Signal` had some way to inspect whether it was `write-able`. Right now you are forced to `try` / `except`. The following API was floated in discussion but I realize it never made it in to an issue:

`Signal._ophyd_readable`
`Signal._ophyd_writeable`

You can guess by checking classes but this approach does not scale well.

## Motivation
Running into problems with `Typhon` if you hand me an arbitrary `ophyd.Signal` I have no idea whether to create a "write" widget for it. 

## Taking things too far
You could apply the same idea to devices `_ophyd_setable`, `ophyd_flyable`. This would allow you to have some sort of check a plan works with the provided devices before execution. Sounds kind of complicated without a huge payoff but just wanted to spit it out.

danielballan, 2018-06-15:
I think we need to do _something_ like this and need to think about in what ways this feature might evolve in the future.

klauer, 2018-11-07:
#630 may or may not address this:

Knowing whether a `Signal` is writable requires an instance rather than the class - checking `read_access` and `write_access` or subscribing to the metadata. Is that sufficient for typhon's purposes?

teddyrendahl, 2018-11-07:
I think it does in part. It looks like `Signal` will be inspectable but not `Device`. These are separate issue but it would be nice if the API wasn't too different.

klauer, 2018-11-08:
At the `Device` level, things get a bit murkier between `set()` and `put()` and what write-ability even means.

Would `write_access` for a `Device` be just `all(signal.write_access for signal in device.all_descendents)`? 

Or `hasattr(device, 'set') and device_promises_it_will_not_raise_readonlyerror`?

teddyrendahl, 2018-11-08:
Not sure of implementation, but I think `ophyd.Device` always has the full interface attached, so `hasattr` won't help you. This has the same issue for the user as a read-only signal. The only way to find which interface the device supports is to look at source code or run full speed into the wall. Can I put this device in a scan as a motor? Can I fly this device?

I could even imagine a `bluesky` preprocessor that inspects your plan and returns a nice error message if you are giving a device a command it can't handle.

ZLLentz, 2018-11-08:
I think the device writability states should be dead simple class attributes. It should be clear when you write your device class whether or not you expect people to call `set`, `kickoff`, etc.

I imagine that we just need a convention for attributes like `settable`, `flyable`, etc. that can default to `False` in the base `Device`, and then be set to `True` in a subclass when applicable. For example, `PositionerBase` would have `settable=True`. It should be up to the device class designer to communicate to the user what each device can and cannot do.