bluesky/ophyd #576: Child Attribute Selection

teddyrendahl, 2018-06-18:
Having some issues with selecting `read_attrs` of a child component. Here is some code to reproduce what I'm seeing. This gives me inconsistent results:

```python
from ophyd import Device, Component as Cpt 
from ophyd.sim import SynAxis

class MyDevice(Device):
    x = Cpt(SynAxis)
    y = Cpt(SynAxis)
    z = Cpt(SynAxis)

    _default_read_attrs = ['x.readback', 'y.readback', 'z.readback',
                           'x.setpoint', 'y.setpoint', 'z.setpoint']

print(MyDevice(name='test').read_attrs)
```
Output:
```bash
$ python blah.py 
['x', 'x.readback', 'x.setpoint', 'y', 'y.readback', 'z', 'z.setpoint']
$ python blah.py 
['x', 'x.setpoint', 'y', 'y.readback', 'y.setpoint', 'z', 'z.setpoint']
$ python blah.py 
['x', 'x.readback', 'x.setpoint', 'y', 'y.setpoint', 'z', 'z.setpoint']
```

It seems like things are getting confused with the redundant naming of a motor and its signal. This also raises the question on what the best way to indicate that you want a `read_attr` that is component of a compnent using the `Kind` syntax.

tacaswell, 2018-06-18:
Well, that is deeply wrong...

I suspect the bug is in here

https://github.com/NSLS-II/ophyd/blob/07ab0b2cfc9fa3c5c35989fd58068df686100304/ophyd/device.py#L834-L844

teddyrendahl, 2018-09-05:
Has this been solved?