bluesky/ophyd #577: Proposal: Add 'interest' level alongside Kind.

danielballan, 2018-06-22:
I am seeking feedback on a proposal that arose in discussion with @ke-zhang-rd and @tacaswell yesterday. Users have requested more individual control over how document-consumers (LiveTable, LivePlot, PeakStats, etc.) interact with hints. For example, they might want several things displayed in LiveTable but only the most interesting couple things displayed in LivePlot.

For the sake of argument, if we gave them _total_ control over what goes where (`plot_hints`, `table_hints`, `peak_stats_hints`, ...) that would defeat the purpose of hints. In that limit, they might as well just define their own custom callbacks -- which of course they can still do. The purpose of hints is to provide an intermediate level of control requiring less specification by the user.

The proposal, then, is this: add an `interest` attribute to every Signal and Device. It will be set to a float between 0 and 1. Update `hints` to look like this:

```py3
>>> eiger1m.hints
{'interest': {'eiger1m_stats5_total': 0.5, 'eiger1m_stats1_total': 1}}
>>> eiger1m.stats5.total.interest = 0.6
>>> eiger1m.hints
{'interest': {'eiger1m_stats5_total': 0.6, 'eiger1m_stats1_total': 1}}
>>> eiger1m.stats5.total.kind = 'normal'
>>> eiger1m.hints  # now stats5 does not appear...
{'interest': {'eiger1m_stats1_total': 1}}
>>> eiger1m.stats5.total.interest  # ... but its interest level is still stored for later...
0.6
>>> eiger1m.stats5.total.kind = 'hinted'  # ...so it can be switched back on
>>> eiger1m.hints
{'interest': {'eiger1m_stats5_total': 0.6, 'eiger1m_stats1_total': 1}}
```

This would be an API change, but a relatively painless one now that `dev.hints` is not directly configurable by the user as of v1.2.0.  And of course`{'fields': list(dev.hints['interest'])}` obtains the original representation.

With this in place, the consumers (i.e. BestEffortCallback or others) can impose a configurable "interest threshold". For example, if I want wide tables but only a couple plots I might do:

```
bec.plots.interest_thresh = 0.9
bec.table.interest_thresh = 0.1
```

Any hinted field with an interest level meeting the threshold will be used.

Are people generally supportive of this idea? There are details to be worked out, such as now interest propagates over the tree.

awalter-bnl, 2018-06-22:
So, I understand the reason for this approach, however I am concerned that this new API is fairly confusing. I think it abstracts from what is trying to be done to much. 

I think it is much better to consider something along the lines of:
`>>>eiger1m.stats5.total.bec
['plot', 'table',......]`
 where if it is hinted it is included only in the bec items listed.
I think this might be something that is easier to follow instead of assigning a 'threshold', and then having to remember what items are set to what threshold prior to setting them.

I also think it is a little easier change to the API than what is suggested.

tacaswell, 2018-06-22:
What would be in the documents in that case @awalter-bnl ? 

If we bake 'bec' into the documents, would we also start baking information about other callbacks in too? 

One advantage of the 'interest' level is that when re-processing the data you could push the stream through with varying threshold on BEC (imagine a thumb-nail generator that only uses the highest interest that presents as a web page that when you click in gives you a table + plots at a 'middle' interest and a 'more' link that takes you to a page with BEC run with a threshold of 0 or a GUI with a couple of sliders that auto re-generates the plots/hides columns of tables).

danielballan, 2018-06-22:
The problem we are trying to solve with "hints" is more generic than supporting `bec` and its few components: this information is for the support of _all_ consumers so it is necessarily abstract. We're thinking about yet-to-be-written visualization tools, data processing pipelines, etc. We definitely don't want to hardcode the names of specific callbacks on the ophyd side.

I think the right place to put the API you want is on the consumer side, where you can specify exactly what you want to see. We should provide them with a consumer-side interface (probably a GUI) to do that.

The hints feature is supposed to help _generic_ consumers figure out what it is interesting in the absence of specific user input. If users start fiddling with interest thresholds in order to add or remove things from plots interactively, I fully agree that would be a terribly annoying API to use. That's not the goal here.

awalter-bnl, 2018-06-22:
I hadn't considered that this could be extended to other consumers, so both @tacaswell and @danielballan are correct in saying that it is by design generic. I also didn't consider that we would be coding in bluesky specific callbacks in ophyd, which I agree we should avoid. So I don;t think my previous suggestions is a good one at present.

> if we gave them total control over what goes where (plot_hints, table_hints, peak_stats_hints, ...) that would defeat the purpose of hints.

I don't completely agree with this statement, but this 'total' control is exactly what is being asked for, as stated in. 

> Users have requested more individual control over how document-consumers (LiveTable, LivePlot, PeakStats, etc.) interact with hints. For example, they might want several things displayed in LiveTable but only the most interesting couple things displayed in LivePlot.

> The purpose of hints is to provide an intermediate level of control requiring less specification by the user.

Is what appears important to me here, what is being suggested is a complicated API that attempts to create a 'middle' ground between giving complete control and just using a single set of 'hints' for all callbacks. My fear here is that this new API will be so complicated to use that it will be just as easy to write custom callbacks that do exactly what is requested. If this is the case then is the really any benefit to it?

teddyrendahl, 2018-06-22:
I am 👎 . Very much see the idea behind this but don't think this is the prefer-able implementation at least from my perspective. I very much like the categorization (hints, normal ...), but applying floating point numbers interest levels seems too obtuse. I also would worry about keeping "interest" level meaning across my device libraries consistent. (I understand you could probably set these at runtimes but still). I'm just not sure how to explain to someone what an interest level of "0.6" means. If users find that making their own callbacks is too much of a burden, I'm not sure managing a bunch of different attribute on an arbitrary 0 to 1 scale is any simpler.

I'll think about this more. I respect the fact that solving this on the device side solves it for all consumer,  but I think this might be avoiding an issue that could be solved by improved callback configuration. Being able to select the values I'm plotting during a scan by clicking will be very intuitive for users, and also helpful for debugging. I could imagine wanting to watch the "ACQUIRE" PV on my AreaDetector without cranking up the "interest" level of the PV e.t.c.

danielballan, 2018-06-22:
> I'm just not sure how to explain to someone what an interest level of "0.6" means.

That is fair.

One possible way forward is to make the live-configurable consumers first and then work backward to consider what additional information we wish they had. Broadly speaking, that general approach has served us well in the past.

klauer, 2018-06-26:
I'm in agreement with @teddyrendahl on this one.

tacaswell, 2018-06-28:
I see a pretty strong analogy to 'zorder' in graphics programs (but I may have a very parochial view of that).

In addition to absolute thresholds you could say "plot the 3 most interesting columns".

awalter-bnl, 2018-06-28:
I think @teddyrendahl specified my concerns far better than I did.

klauer, 2019-01-14:
I think this proposal was received fairly poorly and do not see a direction for further discussion. Closing for now.

closed 2019-01-14 by klauer