bluesky/ophyd #579: Possible changes needed for ADCore R3-3

MarkRivers, 2018-06-26:
These lines in plugins.py will need to be changed for ADCore R3-3
    ts_control = C(EpicsSignal, 'TSControl', string=True)
    ts_current_point = C(EpicsSignal, 'TSCurrentPoint')
    ts_num_points = C(EpicsSignal, 'TSNumPoints')

They should be 
    ts_acquire = C(EpicsSignal, 'TS:TSAcquire', string=True)
    ts_current_point = C(EpicsSignal, 'TS:TSCurrentPoint')
    ts_num_points = C(EpicsSignal, 'TS:TSNumPoints')
    ts_acquire_mode = C(EpicsSignal, 'TS:TSAcquireMode')

I don't see where ts_control was actually used?  Previously it was an mbbo record with these values:
   field(ZRST, "Erase/Start")
   field(ONVL, "1")
   field(ONST, "Start")
   field(TWVL, "2")
   field(TWST, "Stop")
   field(THVL, "3")
   field(THST, "Read")

ts_control has been replaced with ts_acquire.  This is a busy with the values Done (0) and Acquire(1).

danielballan, 2018-06-26:
> I don't see where ts_control was actually used?

I don't believe it was used, just exposed.

Thanks for the heads up! The service to our downstream library is very appreciated.

MarkRivers, 2018-06-26:
I understand.  It is used in client code like that which Bruno referenced here;
https://github.com/NSLS-II-FMX/profile_collection/blob/b1206217476442570c9608fa0c4310ee81ac9e1c/startup/95-custom_plans.py#L271

That code is:
```
        yield from bp.mv(
            slt_gap, gap,                     # Move gap to desired position
            slt_ctr, start - move_slack,      # Move slits to the beginning of the motion
            stats.ts_control, "Erase/Start",  # Prepare statistics Time Series
        )
...
        yield from bp.mv(stats.ts_control, "Stop")
```
So the client code would need to be changed to 
```
            stats.ts_acquire, "Acquire",  # Prepare statistics Time Series
...
        yield from bp.mv(stats.ts_acquire, "Done")
```

What is the best way to handle code that depends on versions?  All drivers and plugins now have an ADCoreVersion_RBV PV which can be used to determine which version of ADCore it was built with.  That would apply to all of the standard plugins in that IOC.  Drivers have a DriverVersion_RBV as well.  

Would the version checking need to be done in the client, or can it be handled in the Ophyd plugin stats class?

danielballan, 2018-06-27:
It seems to me that ophyd should be handling this for the client if possible. What is the oldest version of ADCore that supports ADCoreVersion_RBV?

MarkRivers, 2018-06-27:
> What is the oldest version of ADCore that supports ADCoreVersion_RBV?

R2-6.  The absence of the PV would tell you it is R2-5 or older.

danielballan, 2018-06-27:
Thanks. Created issue: https://github.com/NSLS-II/ophyd/issues/580