bluesky/ophyd #585: Multiple DerivedSignal inputs

teddyrendahl, 2018-06-29:
Any thoughts on making `DerivedSignal` slightly more flexible to allow multiple signals in `derived_from`. This would allow things like normalizing one `Signal` by another e.t.c

klauer, 2018-06-29:
👍 It could definitely be useful. I can't think of a reason not to allow it.

tacaswell, 2018-07-01:
This is approaching a pseudo positioner...

danielballan, 2018-07-02:
Could be worth looking at the implementation and deciding whether it should just be PseudoPositioner or is worth having a new thing for.

teddyrendahl, 2018-07-02:
@tacaswell I buy this. 👍 

Next question, slightly unrelated, running into an issue where I'm "read"-ing my `DerivedSignal` but my background signal is not getting triggered. Any thoughts on passing the `trigger` event through?

```python
def trigger(self):
     return self.derived_from.trigger()
```
We can always just leave this up to the subclass, but seems like most applications would want this behavior.

tacaswell, 2018-07-05:
I think we want to have a half-pseudo-positioner class that only provides the `inverse` (that is real -> pseudo) mapping.

I can see this being super useful when handling scalars.