bluesky/ophyd #586: extra - unwanted - information in config_attrs for AD

ambarb, 2018-07-04:
2ID's `rixscam()` class is based on `AreaDetector`.  We now see a lot of stuff in `config_attrs` that we do not want to see.  The new things are associated with hdf5.

ophyd 1.2.0a3
bluseky 1.3.0

`rixscam :   {'rixscam_cam_acquire_period': 10.0, 'rixscam_cam_acquire_time': 2.5, 'rixscam_cam_num_exposures': 1, 'rixscam_cam_temperature': -110.0, 'rixscam_cam_temperature_actual': -104.6329437, 'rixscam_cam_trigger_mode': 2, 'rixscam_hdf5_configuration_names': ['rixscam_hdf5_configuration_names', 'rixscam_hdf5_port_name', 'rixscam_hdf5_asyn_pipeline_config', 'rixscam_hdf5_blocking_callbacks', 'rixscam_hdf5_enable', 'rixscam_hdf5_nd_array_port', 'rixscam_hdf5_plugin_type', 'rixscam_hdf5_auto_increment', 'rixscam_hdf5_auto_save', 'rixscam_hdf5_file_format', 'rixscam_hdf5_file_name', 'rixscam_hdf5_file_path', 'rixscam_hdf5_file_path_exists', 'rixscam_hdf5_file_template', 'rixscam_hdf5_file_write_mode', 'rixscam_hdf5_full_file_name', 'rixscam_hdf5_num_capture', 'rixscam_hdf5_boundary_align', 'rixscam_hdf5_boundary_threshold', 'rixscam_hdf5_compression', 'rixscam_hdf5_data_bits_offset', 'rixscam_hdf5_extra_dim_name', 'rixscam_hdf5_extra_dim_size', 'rixscam_hdf5_io_speed', 'rixscam_hdf5_num_col_chunks', 'rixscam_hdf5_num_data_bits', 'rixscam_hdf5_num_extra_dims', 'rixscam_hdf5_num_frames_chunks', 'rixscam_hdf5_num_frames_flush', 'rixscam_hdf5_num_row_chunks', 'rixscam_hdf5_run_time', 'rixscam_hdf5_szip_num_pixels', 'rixscam_hdf5_store_attr', 'rixscam_hdf5_store_perform', 'rixscam_hdf5_zlevel'], 'rixscam_hdf5_port_name': 'FileHDF1', 'rixscam_hdf5_asyn_pipeline_config': ['rixscam_cam_configuration_names', 'rixscam_hdf5_configuration_names'], 'rixscam_hdf5_blocking_callbacks': 'Yes', 'rixscam_hdf5_enable': 'Enable', 'rixscam_hdf5_nd_array_port': 'cam1', 'rixscam_hdf5_plugin_type': 'NDFileHDF5 ver1.8.14', 'rixscam_hdf5_auto_increment': 1, 'rixscam_hdf5_auto_save': 1, 'rixscam_hdf5_file_format': 0, 'rixscam_hdf5_file_name': 'e64f5039-6c9b-42b8-9879', 'rixscam_hdf5_file_path': 'X:\\RIXSCAM\\DATA\\2018\\07\\03\\', 'rixscam_hdf5_file_path_exists': 1, 'rixscam_hdf5_file_template': '%s%s_%6.6d.h5', 'rixscam_hdf5_file_write_mode': 2, 'rixscam_hdf5_full_file_name': 'X:\\RIXSCAM\\DATA\\2018\\07\\03\\e64f5039-6c9b-42b8-9879_000000.h5', 'rixscam_hdf5_num_capture': 0, 'rixscam_hdf5_boundary_align': 0, 'rixscam_hdf5_boundary_threshold': 65536, 'rixscam_hdf5_compression': 0, 'rixscam_hdf5_data_bits_offset': 0, 'rixscam_hdf5_io_speed': 20.91848527103553, 'rixscam_hdf5_num_col_chunks': 6604, 'rixscam_hdf5_num_data_bits': 8, 'rixscam_hdf5_num_extra_dims': 0, 'rixscam_hdf5_num_frames_chunks': 1, 'rixscam_hdf5_num_frames_flush': 0, 'rixscam_hdf5_num_row_chunks': 1612, 'rixscam_hdf5_run_time': 23.296098426, 'rixscam_hdf5_szip_num_pixels': 16, 'rixscam_hdf5_store_attr': 1, 'rixscam_hdf5_store_perform': 1, 'rixscam_hdf5_zlevel': 6, 'rixscam_set_node': 1, 'rixscam_sensor_region_xsize': 3300, 'rixscam_sensor_region_ysize': 1610, 'rixscam_sensor_region_xstart': 0, 'rixscam_sensor_region_ystart': 0, 'rixscam_sensor_binning_x': 1, 'rixscam_sensor_binning_y': 1, 'rixscam_ccd1_hv': 20.0, 'rixscam_ccd2_hv': 20.0} `

## last cycle's startup file for the rixscam
https://github.com/NSLS-II-SIX/profile_collection/blob/master/startup/22-rixscam.py 

## current configuration
```python
rixscam.configuration_attrs = ['cam.acquire_time', 'cam.acquire_period',
                               'cam.num_exposures',
                               'cam.temperature', 'cam.temperature_actual',
                               'cam.trigger_mode', 'ccd1_hv', 'ccd2_hv', 
                               'set_node',
                               #'sensor_xsize', 'sensor_ysize',
                               'sensor_region_xsize', 'sensor_region_ysize', 
                               'sensor_region_xstart', 'sensor_region_ystart', 
                               'sensor_binning_x', 'sensor_binning_y'] 
```

```shell
abarbour@xf02id1-ws2:/home/xf02id1/.ipython/profile_collection/startup$ grep -Rl rixscam *
22-rixscam.py
```

ambarb, 2018-07-04:
have not check proslica class yet.  We can do this when we have time here.

ambarb, 2018-07-04:
`hdr.config_data('diagon_h_cam')['primary'][0]`

`{'diagon_h_cam_cam_acquire_period': 1.0,
 'diagon_h_cam_cam_acquire_time': 0.1,
 'diagon_h_cam_cam_image_mode': 1,
 'diagon_h_cam_cam_manufacturer': 'Prosilica',
 'diagon_h_cam_cam_model': 'Mako G-125B',
 'diagon_h_cam_cam_num_exposures': 1,
 'diagon_h_cam_cam_trigger_mode': 5}`

awalter-bnl, 2018-07-05:
Hi Andi, is the hdf5 plugin enabled for 'diagon_h_cam'? if not can you see what the outcome is with it enabled? This will tell us if the problem is the rixscam class or the area-detector hdf5 plugin.

ambarb, 2018-07-05:
will put on the to do list for the next beam dump

tacaswell, 2018-07-06:
This is likely due to the change read configuration from any child we read from (which seems like a reasonable position so we don't accidentally forget to write down things we need), however this is exposing miss-configuration where AD plugins have way too many things marked as configurationation.  The fix is to change the kind on the components you don't want to see to `'omitted'`.  This should probably be done at the ophyd level (so the defaults are useful).

closed 2025-11-22 by ambarb