bluesky/ophyd #587: extra - unwanted - information in config_attrs for qem  (Quad-EM)

ambarb, 2018-07-07:
I know this is related to #586 but just wanted to document that it is not just hdf5 stuff for AreaDetector.

Scans before didn't have a bunch of histogram data in the config_attrs.  Now, functions do not work because there is extra stuff in a weird format.

```python
def scan_dets(*scan_no):
    '''Enter scan id numbers to print settings for primary stream detectors'''
    xtr_str = ''
    for scan in scan_no:
        if 'reason' in db[scan].start:
            xtr_str = '   ' + db[scan].start['reason'] 
        elif 'purpose' in db[scan].start:
            xtr_str = '   ' + db[scan].start['purpose']
        else:
            xtr_str = '   '
        print('Scan ID {}:  {}'.format(scan, xtr_str) )
        
        for detector in db[scan]['start']['detectors']:
            print('  {} (primary stream):  '.format(detector) )
            settings = db[scan].config_data(detector)['primary'][0]
            for p, n in settings.items():
                print('    {:_<30} : {:_>20}'.format(p, n))
            print('{:-<70}'.format('-'))

dets = scan_dets(12464, 12465,12467, 17990)

Scan ID 12467:     test scan
  qem07 (primary stream):  
    qem07_integration_time________ : ______________0.0004
    qem07_averaging_time__________ : ________________0.01
    qem07_em_range________________ : _______________12 pC
    qem07_num_averaged____________ : __________________13
    qem07_values_per_read_________ : ___________________1
----------------------------------------------------------------------
Scan ID 17990:     
  qem07 (primary stream):  
    qem07_integration_time________ : ______________0.0004
    qem07_averaging_time__________ : _________________1.0
    qem07_em_range________________ : _______________12 pC
    qem07_values_per_read_________ : ___________________1
    qem07_num_averaged____________ : ________________1250
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-17-e3e0af5cd157> in <module>()
----> 1 dets = scan_dets(12464, 12465,12467, 17990)

<ipython-input-15-02346f68bca2> in scan_dets(*scan_no)
     15             settings = db[scan].config_data(detector)['primary'][0]
     16             for p, n in settings.items():
---> 17                 print('    {:_<30} : {:_>20}'.format(p, n))
     18             print('{:-<70}'.format('-'))

TypeError: unsupported format string passed to list.__format__
```

# NOW
```
db[17990].config_data('qem07')['primary'][0]

{'qem07_averaging_time': 1.0,
 'qem07_current1_asyn_pipeline_config': ['qem07_conf_configuration_names',
  'qem07_current1_configuration_names'],
 'qem07_current1_bgd_width': 1,
 'qem07_current1_blocking_callbacks': 'Yes',
 'qem07_current1_centroid_threshold': 1.0,
 'qem07_current1_compute_centroid': 'No',
 'qem07_current1_compute_histogram': 'No',
 'qem07_current1_compute_profiles': 'No',
 'qem07_current1_compute_statistics': 'Yes',
 'qem07_current1_configuration_names': ['qem07_current1_configuration_names',
  'qem07_current1_port_name',
  'qem07_current1_asyn_pipeline_config',
  'qem07_current1_blocking_callbacks',
  'qem07_current1_enable',
  'qem07_current1_nd_array_port',
  'qem07_current1_plugin_type',
  'qem07_current1_bgd_width',
  'qem07_current1_centroid_threshold',
  'qem07_current1_compute_centroid',
  'qem07_current1_compute_histogram',
  'qem07_current1_compute_profiles',
  'qem07_current1_compute_statistics',
  'qem07_current1_hist_max',
  'qem07_current1_hist_min',
  'qem07_current1_hist_size',
  'qem07_current1_profile_cursor',
  'qem07_current1_profile_size',
  'qem07_current1_ts_control',
  'qem07_current1_ts_num_points'],
 'qem07_current1_enable': 'Enable',
 'qem07_current1_hist_max': 255.0,
 'qem07_current1_hist_min': 0.0,
 'qem07_current1_hist_size': 256,
 'qem07_current1_nd_array_port': 'EM180',
 'qem07_current1_plugin_type': 'NDPluginStats',
 'qem07_current1_port_name': 'STATS1',
 'qem07_current1_ts_control': 'Erase/Start',
 'qem07_current1_ts_num_points': 1000,
 'qem07_current2_asyn_pipeline_config': ['qem07_conf_configuration_names',
  'qem07_current2_configuration_names'],
 'qem07_current2_bgd_width': 1,
 'qem07_current2_blocking_callbacks': 'Yes',
 'qem07_current2_centroid_threshold': 1.0,
 'qem07_current2_compute_centroid': 'No',
 'qem07_current2_compute_histogram': 'No',
 'qem07_current2_compute_profiles': 'No',
 'qem07_current2_compute_statistics': 'Yes',
 'qem07_current2_configuration_names': ['qem07_current2_configuration_names',
  'qem07_current2_port_name',
  'qem07_current2_asyn_pipeline_config',
  'qem07_current2_blocking_callbacks',
  'qem07_current2_enable',
  'qem07_current2_nd_array_port',
  'qem07_current2_plugin_type',
  'qem07_current2_bgd_width',
  'qem07_current2_centroid_threshold',
  'qem07_current2_compute_centroid',
  'qem07_current2_compute_histogram',
  'qem07_current2_compute_profiles',
  'qem07_current2_compute_statistics',
  'qem07_current2_hist_max',
  'qem07_current2_hist_min',
  'qem07_current2_hist_size',
  'qem07_current2_profile_cursor',
  'qem07_current2_profile_size',
  'qem07_current2_ts_control',
  'qem07_current2_ts_num_points'],
 'qem07_current2_enable': 'Enable',
 'qem07_current2_hist_max': 255.0,
 'qem07_current2_hist_min': 0.0,
 'qem07_current2_hist_size': 256,
 'qem07_current2_nd_array_port': 'EM180',
 'qem07_current2_plugin_type': 'NDPluginStats',
 'qem07_current2_port_name': 'STATS2',
 'qem07_current2_ts_control': 'Erase/Start',
 'qem07_current2_ts_num_points': 1000,
 'qem07_current3_asyn_pipeline_config': ['qem07_conf_configuration_names',
  'qem07_current3_configuration_names'],
 'qem07_current3_bgd_width': 1,
 'qem07_current3_blocking_callbacks': 'Yes',
 'qem07_current3_centroid_threshold': 1.0,
 'qem07_current3_compute_centroid': 'No',
 'qem07_current3_compute_histogram': 'No',
 'qem07_current3_compute_profiles': 'No',
 'qem07_current3_compute_statistics': 'Yes',
 'qem07_current3_configuration_names': ['qem07_current3_configuration_names',
  'qem07_current3_port_name',
  'qem07_current3_asyn_pipeline_config',
  'qem07_current3_blocking_callbacks',
  'qem07_current3_enable',
  'qem07_current3_nd_array_port',
  'qem07_current3_plugin_type',
  'qem07_current3_bgd_width',
  'qem07_current3_centroid_threshold',
  'qem07_current3_compute_centroid',
  'qem07_current3_compute_histogram',
  'qem07_current3_compute_profiles',
  'qem07_current3_compute_statistics',
  'qem07_current3_hist_max',
  'qem07_current3_hist_min',
  'qem07_current3_hist_size',
  'qem07_current3_profile_cursor',
  'qem07_current3_profile_size',
  'qem07_current3_ts_control',
  'qem07_current3_ts_num_points'],
 'qem07_current3_enable': 'Enable',
 'qem07_current3_hist_max': 255.0,
 'qem07_current3_hist_min': 0.0,
 'qem07_current3_hist_size': 256,
 'qem07_current3_nd_array_port': 'EM180',
 'qem07_current3_plugin_type': 'NDPluginStats',
 'qem07_current3_port_name': 'STATS3',
 'qem07_current3_ts_control': 'Erase/Start',
 'qem07_current3_ts_num_points': 1000,
 'qem07_current4_asyn_pipeline_config': ['qem07_conf_configuration_names',
  'qem07_current4_configuration_names'],
 'qem07_current4_bgd_width': 1,
 'qem07_current4_blocking_callbacks': 'Yes',
 'qem07_current4_centroid_threshold': 1.0,
 'qem07_current4_compute_centroid': 'No',
 'qem07_current4_compute_histogram': 'No',
 'qem07_current4_compute_profiles': 'No',
 'qem07_current4_compute_statistics': 'Yes',
 'qem07_current4_configuration_names': ['qem07_current4_configuration_names',
  'qem07_current4_port_name',
  'qem07_current4_asyn_pipeline_config',
  'qem07_current4_blocking_callbacks',
  'qem07_current4_enable',
  'qem07_current4_nd_array_port',
  'qem07_current4_plugin_type',
  'qem07_current4_bgd_width',
  'qem07_current4_centroid_threshold',
  'qem07_current4_compute_centroid',
  'qem07_current4_compute_histogram',
  'qem07_current4_compute_profiles',
  'qem07_current4_compute_statistics',
  'qem07_current4_hist_max',
  'qem07_current4_hist_min',
  'qem07_current4_hist_size',
  'qem07_current4_profile_cursor',
  'qem07_current4_profile_size',
  'qem07_current4_ts_control',
  'qem07_current4_ts_num_points'],
 'qem07_current4_enable': 'Enable',
 'qem07_current4_hist_max': 255.0,
 'qem07_current4_hist_min': 0.0,
 'qem07_current4_hist_size': 256,
 'qem07_current4_nd_array_port': 'EM180',
 'qem07_current4_plugin_type': 'NDPluginStats',
 'qem07_current4_port_name': 'STATS4',
 'qem07_current4_ts_control': 'Erase/Start',
 'qem07_current4_ts_num_points': 1000,
 'qem07_em_range': '12 pC',
 'qem07_integration_time': 0.0004,
 'qem07_num_averaged': 1250,
 'qem07_values_per_read': 1}
```

awalter-bnl, 2018-07-07:
Thanks for the heads up Andi, this is the same problem. The new lazy loading of qem class also has the same problem as either the hdf5 plugin for area detectors or possibly the RIXS cam class it appears. In that there a whole heap more config attrs than previous and we will need to go back and remove a lot of them from config attrs.

In regard to the functions, personally I think you should consider modifying them. It appears that your current function assumes that the value associated with an attribute is a float, ophyd does not make this assumption so it can (and probably will) become a problem later when someone adds a non- float value to an attribute. There are in fact already many attrs that have non float values that could cause issues. Probably the best option is to use a try, except here that try’s to round but if it throws a type error just prints the value.

ambarb, 2018-07-07:
Indeed.  As soon I entered the information above, I added `try` and `except`.  I am just passing on the exception.

Maybe I am missing something, but I did not explicitly declare `:f`  is that inherent to the method of declaring characters? 

The exception that I get I think is related to the next key being assigned a value that is a list of strings, which isn't a simple output to deal with and I am not sure if that is something that I would ever want to see.

Maybe this extra info can be given a key that is universal to more lower level things associated with writing files and centroiding.

awalter-bnl, 2018-07-07:
Great, I will make a note that we need to look at the number of config attrs on qem’s and area detectors.

You are correct in saying that it is not assuming a float, but a float or string which can be formatted. Printing without the formatting will work with anything else, but it could turn out to be a lot of characters. If you are only interested in config attrs that are floats or strings you can pass on except as you say.

In regard to considering only allowing floats and strings as the output for attrs. This would break far too much to be a realistic option. In addition there are several well defined cases when a string or a float is not applicable. One in particular that we have discussed is the “regions” associated with images and dark areas from the raw image. As I mentioned I would like this to be saved via a config attr on the RIXS camera. This I think is best saved as a dictionary, but this we can discuss on Monday (perhaps in the software meeting).

tacaswell, 2018-07-09:
Sorry this is being painful, but I am pretty excited that someone is looking at the configuration information !

ambarb, 2018-07-10:
The discussion of the code that access configuration information has moved to https://github.com/NSLS-II-SIX/sixtools/pull/2#pullrequestreview-135766276 

However, with all the information that is currently available for AD rixscam and qem, I think it is less useful - and in fact distracting - to print the entire configuration dictionary.  I am requesting special dictionaries for the file writing,  roi processing and so on in an effort to have a universal `pass` unless someone wants a verbose view.