bluesky/ophyd #588: Proposal: Signal Registry

teddyrendahl, 2018-07-09:
## Basic Idea
In the last few weeks I have had multiple discussions that have come to the conclusion it would be very nice to have an `ophyd.Signal` registry. This would essentially be a weak reference map of signal name to signal so that you can search all the signals that have been created in your Python session. When you create a new signal it gets added to the Registry and when it is garbage collected it disappears. 

## Applications
I think there are a number of questions that are difficult to answer in the current schema without tracking down all the `ophyd` objects in the current scope. The `SignalRegistry` would make these simple to answer:
* How many connections do I currently have open?
* I control this PV with my GUI, is there any Ophyd device to control this?

There are also a few future thinking reasons I think this could be useful

### Typhon
Currently `typhon` has it's own version of the signal registry. This is because when you create a string identifier for your channel i.e `"sig://my_motor.setpoint"` you don't want to create a new signal, but searching around for the existing one in a namespace is cumbersome. Having this implemented in `Ophyd` would spare users from explicitly calling "register_signal" which is cumbersome...

### Caproto
From my understanding `caproto` bench testing has shown huge performance increases by batching connections. Collecting signals then connecting them all at once in this manner could significantly reduce the overhead of loading a bunch of different devices. 

Just some thoughts, curious what people think. Inspired by varying discussions with @hhslepicka, @ZLLentz and @ronpandolfi

danielballan, 2018-07-09:
I am in favor of trying to do this. I think one point of discussion will be, do `Signal` objects automatically register themselves on `__init__` or do they have to be separately registered? In previous iterations of a similar discussion, this has been a sticking point.

tacaswell, 2018-07-09:
The 'bunch connections' issue can probably be solved by tweaking how `connect` on device works (it is responsible for collecting all of it's children's PVs to find and then distributing them back out modulo not baking EPICS into `Device`).

By 'signal name' here do you mean the PV (not `sig.name`)?  If so, this should probably be put in the ControlLayer and  hi-jacks the call to `get_pv` to keep a mapping between a PV string and a list of weak-refs to ophyd objects that have called `get_pv` with that value (we do not _forbid_ creating more than one EpicsSignal looking at the same underlying PV, it is probably a bad idea, but you can do it). There is currently a flag on `set_cl` that enables a very weak version of this for debugging purposes https://github.com/NSLS-II/ophyd/blob/f4fdae506ac33b1611471eb7f5a406c3b39e1f0c/ophyd/__init__.py#L45-L57 
which just gives you the count of how many times you tried to connect to a given PV.

> I control this PV with my GUI, is there any Ophyd device to control this?

I am not convinced that we want to conjure this information up, it would be better to explicitly keep a mapping, either by constructing one from the other or both from the same ground source of truth.

teddyrendahl, 2018-07-09:
@danielballan I did imagine doing it on `__init__`. Explicitly having to add things to the cache seems kind of self-defeating. I also don't see too much harm in keeping an internal record even if a user never plans to use them.

@tacaswell 
> By 'signal name' here do you mean the PV (not sig.name)? 

No, I would like this solution to be independent of signal type. Typhon wants any signal it is going to communicate with to be in a registry irregardless of class / control layer. I understand that signal names are not guaranteed to be unique so some thought has to be given to what we can use as the keys.

klauer, 2018-07-09:
@dchabot I think we should implement a signal registry - let's call it "session manager" - to keep track of signals. What do you think? (sorry, I couldn't help myself...)

dchabot, 2018-07-09:
@klauer 
Rediscovery is neat! Lol

tacaswell, 2018-07-10:
> I understand that signal names are not guaranteed to be unique so some thought has to be given to what we can use as the keys.

Signal names are also run-time mutable.

If I do

```python
A = EpicsSignal('foo', name='A')
B = EpicsSignal('foo', name='B')
C = EpicsSignal('foo', name='A')
D = EpicsSignal('bar', name='B')
E = EpicsSginal('foo', name='E')
E.name = 'A'
C.name = 'C'
```

what do you put in the registry and when?

For epics based signals there is something unique to hold on to (the PV), if I do

```python
a1 = Signal(name='a')
a2 = Signal(name='a')
```

what is the right thing to do?

To do this reliably I think you need to do it (at least) one layer above ophyd where you can make stronger guarantees that silliness like the above does not happen.

ronpandolfi, 2018-07-10:
What is the justification for signal names being mutable? If they are made immutable you could use the name as the registry key?

ZLLentz, 2018-07-11:
Is it necessary for this registry to have a mapping at all? It's not obvious to me that a name-based mapping is useful for these applications, even if there wasn't a potential for collisions. A really basic `WeakSet` of all the signal objects that I can iterate though would be extremely useful on its own, as it saves me the hassle of crawling through my python namespace.

Then, using this basic set as a starting point, an application using `ophyd` could choose to construct a PV mapping of epics signals if you want to search for devices based on PVs, or a name mapping, or really any application-specific grouping that makes sense in that context.

tacaswell, 2018-07-11:
> What is the justification for signal names being mutable? If they are made immutable you could use the name as the registry key?

`obj.name` is what shows up in the dictionary via `read` which is something BL staff want / need to be able to control.  If it is not run time mutable we need to find a way to thread the name of every child through the device init.

> ...and then by metaclassing you could have a1 and a2 being the same object?

I do not think it is clear that you do want `a1` and `a2` to be the same object.  For example

```python
group1 = SimpleNamespace(x=EpicsMotor('stage1:x', name='x'), y=EpicsMotor('stage1:y', name='y'), a=Signal(name='a'))
group2 = SimpleNamespace(x=EpicsMotor('stage:2:x', name='x'), y=EpicsMotor('stage2:y', name='y'), a=Signal(name='a'))
```

where the groups are set up such that you will only ever want to use group1 or group2 simultaneously but also need the ability to switch between the two sets.

 

> really any application-specific grouping

But again, it seems simpler to have explicitly opt-into the application logic rather than try to infer what that structure was.