bluesky/ophyd #591: ScalerCH needs some documentation

prjemian, 2018-07-25:
There is zero documentation about this device.  It's an improvement over `EpicsScaler`, right?  Ready for production use?

danielballan, 2018-07-30:
> There is zero documentation about this device.

Ugh, sorry. I'll work on that and use this issue to track it.

> It's an improvement over EpicsScaler, right? 

Yes. The layout of the components is more natural, in our opinion.

> Ready for production use?

Yes, used in production at multiple NSLS-II beamlines.

danielballan, 2018-07-30:
@prjemian To tide you over, an example in production: https://github.com/NSLS-II-TES/profile_collection/blob/master/startup/25-sclr.py

tacaswell, 2018-07-31:
Scalers present a particularly challenging surface for ophyd.  It seems to be that the MCA mode and the 'normal counting' mode are more-or-less orthogonal sets of controls that talk to the same underlying hardware (hence the sorta-hacky 'set_mode' stuff).  You want to make the separate devices because their logic / PVs / etc are separate, but the same device to ensure you do not try to use both 'views' in the same scan.

prjemian, 2018-07-31:
Thanks.

APS USAXS will use [`ScalerCH`](https://github.com/APS-USAXS/ipython-usaxs/issues/37)

Here's a tutorial for `ScalerCH`, written for the absolute novice (who conveniently happens to have a synApps 5.8 scaler record nearby):
https://github.com/BCDA-APS/use_bluesky/blob/master/notebooks/lesson1.ipynb

prjemian, 2018-07-31:
USAXS: https://github.com/APS-USAXS/ipython-usaxs/blob/master/profile_bluesky/startup/20-detectors.py#L10

PR #1005 (2021-08-24): DOC #591 add ScalerCH docs
Fix #591 (from 2018) and finally reveal the `ScalerCH` class in the documentation.  Also update the URLs to the EPICS synApps `scaler`, since in the interim, it has been moved out of the synApps `std` module.
closed 2021-08-24 by mrakitin