bluesky/ophyd #598: TimeoutError when area detector is just rebooted

prjemian, 2018-08-17:
This is a consistent pattern:

* start bluesky (ipython, notebook, whatever)
* connect with area detector AND its HDF5 file writer plugin
* reboot area detector IOC
* try areadetector.trigger()

This will fail with `TimeoutError` on the `$(P):HDF1:Capture_RBV`

The fix for this is to always, *first* after rebooting the IOC, acquire a frame and save it from the HDF5 file writer plugin.

prjemian, 2018-08-17:
Here's the exception trace from today:

```
[user2bmb@lyra,42,startup]$ use_bluesky.sh 2bmb
BlueSky executable path: /home/beams/USER2BMB/Apps/BlueSky/bin
Python 3.6.5 :: Anaconda, Inc.
conda 4.5.8
pip 10.0.1 from /home/beams/USER2BMB/Apps/BlueSky/lib/python3.6/site-packages/pip (python 3.6)
starting ipython with profile: 2bmb
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: 2bmb
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/00-0-checks.py
BlueSky version: 1.3.3
Ophyd version: 1.2.0
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/00-startup.py
Loading metadata history from /home/beams/USER2BMB/.config/bluesky/bluesky_history.db
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/01-databroker.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/02-pyepics.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/10-imports.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/11-motors.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/15-custom-devices.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/20-signals.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/25-PG3-grasshopper.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/30-busy_fly_scan.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/50-plans.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/60-handler.py
/home/beams8/USER2BMB/.ipython/profile_2bmb/startup/60-metadata.py
using profile: <property object at 0x7f05bbf1e9a8>
Metadata dictionary:
RE.md['APS_BlueSky_tools_VERSION'] = 0.0.22
RE.md['BLUESKY_VERSION'] = 1.3.3
RE.md['EPICS_APP'] = /net/s2dserv/xorApps/epics/synApps_5_6/ioc/2bmb
RE.md['EPICS_AR_PORT'] = 7002
RE.md['EPICS_CA_MAX_ARRAY_BYTES'] = 100000000
RE.md['EPICS_EXTENSIONS'] = /APSshare/epics/extensions
RE.md['EPICS_HOST_ARCH'] = linux-x86_64
RE.md['OPHYD_VERSION'] = 1.2.0
RE.md['SESSION_STARTED'] = 2018-08-17 10:08:07.467401
RE.md['USER2BMB_ROOT_DIR'] = /local/user2bmb
RE.md['beamline_id'] = 2-BM tomography
RE.md['ipython_profile'] = <property object at 0x7f05bbf1e9a8>
RE.md['login_id'] = user2bmb@lyra.xray.aps.anl.gov
RE.md['pid'] = 11721
RE.md['profile'] = 2bmb
RE.md['proposal_id'] = 57473
RE.md['scan_id'] = 214

In [1]: RE(user_tomo_scan(), comment="Look for theta in metadata", sample="wood stick")
---------------------------------------------------------------------------
TimeoutError                              Traceback (most recent call last)
<ipython-input-1-acf0ea33e291> in <module>()
----> 1 RE(user_tomo_scan(), comment="Look for theta in metadata", sample="wood stick")

~/Apps/BlueSky/lib/python3.6/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    742                     # it (unless it is a canceled error)
    743                     if exc is not None:
--> 744                         raise exc
    745 
    746             if self._interrupted:

~/Apps/BlueSky/lib/python3.6/site-packages/bluesky/run_engine.py in _run(self)
   1251             self.log.error("Run aborted")
   1252             self.log.error("%r", err)
-> 1253             raise err
   1254         finally:
   1255             # Some done_callbacks may still be alive in other threads.

[... 255 lines cut ...]
~/Apps/BlueSky/lib/python3.6/site-packages/ophyd/areadetector/filestore_mixins.py in stage(self)
    295         self._locked_key_list = False
    296         self._datum_uids.clear()
--> 297         super().stage()
    298         # Subclasses will assemble resource_kwargs and call
    299         # self._generate_resource(resource_kwrags)

~/Apps/BlueSky/lib/python3.6/site-packages/ophyd/device.py in stage(self)
    596                                self.name,
    597                                val, original_vals[sig])
--> 598                 set_and_wait(sig, val)
    599                 # It worked -- now add it to this list of sigs to unstage.
    600                 self._original_vals[sig] = original_vals[sig]

~/Apps/BlueSky/lib/python3.6/site-packages/ophyd/utils/epics_pvs.py in set_and_wait(signal, val, poll_time, timeout, rtol, atol)
    263             raise TimeoutError("Attempted to set %r to value %r and timed "
    264                                "out after %r seconds. Current value is %r." %
--> 265                                (signal, val, timeout, current_value))
    266 
    267 

TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='2bmbPG3:HDF1:Capture_RBV', name='pg3_det_hdf1_capture', parent='pg3_det_hdf1', value=0, timestamp=1534518619.46664, pv_kw={}, auto_monitor=False, string=False, write_pv='2bmbPG3:HDF1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.

In [2]: 
```

danielballan, 2018-08-20:
I'm glad that it's at least repeatable. Have you tried poking around with plain pyepics to see if pyepics can connect to that PV?

tacaswell, 2018-08-21:
Can we tell from the client side if an IOC has been re-booted?

This is probably related to the issue we have seen with the hdf5 plugin where if you do not 'warm it up' by putting a frame through you get segfaults and the issue we were seeing at SRX where we were trying to allocate 0 sized arrays and something at the c++ layer was protesting (not sure if it was hdf5 or AD).

There is a `warmup` method on plugins to do the "take one frame" procedure
<details>

https://github.com/NSLS-II/ophyd/blob/25be779cfbd65750956d2514f917d92eb23ec85a/ophyd/areadetector/plugins.py#L883-L909

</details>

Is this something worth talking to Mark about to see if the plugins could pull some of this information when they are connected?  I suspect not as if something upstream changes size etc, the downstream plugins should not change their meta-data until they see a new array.

tacaswell, 2018-08-21:
On a bit more consideration, is there a PV that tells us if the plugin is ready to go?  Maybe `ExecutionTime_RBV` if the boot-up value is stable?

We are seeing issues similar to this where the IOC seems to restart mid-acquisition and we get this exception on unstage.

hhslepicka, 2018-08-21:
You can keep track of reboot and other connectivity issues by using the
connection callback and keeping the state at the Epics Signal...

prjemian, 2018-08-21:
I thoroughly believe there is a "prime the pump" issue with the HDF5 plugin on first acquisition after IOC startup.  

Issue #598 (this one) is with how bluesky responds to such a problem.  The report of a TimeoutError on the Capture_RBV is a symptom of the same situation but leads to a lot of time wasted looking at the Capture_RBV PV.  Simply doing a "warmup" procedure on the detector is all it takes so this error does not happen.

1. The HDF5 plugin *warmup* problem should be described and resolved.  In the appropriate AreaDetector repo.
2. The TimeoutError could be trapped specifically, since it is a known behavior, and a better message returned.  Something like: "You need to use the HDF5 plugin with at least one image first."

tacaswell, 2018-08-22:
Are we sure that this problem only affects the HDF5 plugin (and not all AD plugins)?

@hhslepicka Will that let us tell apart "lost connection because IOC restarted" from "lost connection because network flaked out"?

prjemian, 2018-08-22:
Right.  I see these as two separate situations.  In *this* case, we can connect with the PV but there is a problem we discover only when we trigger.  In the other case, the PV has been disconnected (or did not connect to start with).

In this case, we should take no action.  With the HDF5 plugin, it is a problem that belongs to the area detector HDF5 plugin software and we find ourselves caught up in it, recognizing that it is a problem, and needing to react based on that understanding.

prjemian, 2018-08-22:
And I'm not sure this is only with the HDF5 plugin.  I rather suspect it is an operational issue with the AD design, that no NDarray has been pushed to the plugin chain at all.

Our DAQ software should not take responsibility to make additional triggers of the detector.  Take the streaming tomography workflow we've been developing, we expect and have prepared the AD modules to trigger and receive a precise number of image frames.  The warmup procedure is well intentioned but can produce a different number of frames than were programmed.  Best, to end the acquisition sequence and set things right before proceeding in the case of our streaming tomography.

danielballan, 2018-08-22:
So do you think the right course here is just to try to detect where the plugin needs to be warmed up and raise with an informative error message? That seems like a noncontroversial initial step (and perhaps it's far as we want to go). We can detect this by checking whether the file plugin --- any file plugin -- knows its ArraySize. The ArraySize initializes to 0 on boot and becomes set when the first image passes through.

prjemian, 2018-08-22:
I agree.

prjemian, 2018-08-22:
Just started my IOC, then started bluesky ipython session.  Have not acquired any images (no NDarray pushed through the plugin chain).

```
```

prjemian, 2018-08-22:
PyEpics and Ophyd can both see the PV reported:

```
In [3]: ! caget otzSIM1:HDF1:Capture_RBV
otzSIM1:HDF1:Capture_RBV       Done

In [4]: pv = epics.PV("otzSIM1:HDF1:Capture_RBV")

In [5]: pv.get()
Out[5]: 0

In [6]: adsimdet.hdf1.capture.read()
Out[6]: {'adsimdet_hdf1_capture': {'value': 0, 'timestamp': 1534974804.299655}}
```

prjemian, 2018-08-22:
The exception occurs on staging:

```
In [7]: adsimdet.hdf1.stage_sigs
Out[7]: 
OrderedDict([('enable', 1),
             ('auto_increment', 'Yes'),
             ('array_counter', 0),
             ('auto_save', 'Yes'),
             ('num_capture', 0),
             ('file_template', '%s%s_%6.6d.h5'),
             ('file_write_mode', 'Stream'),
             ('capture', 1),
             ('blocking_callbacks', 'Yes'),
             ('parent.cam.array_callbacks', 1)])
```

prjemian, 2018-08-22:
This is the operation that does not succeed:

```
In [14]: adsimdet.hdf1.capture.put(1)

In [15]: adsimdet.hdf1.capture.value
Out[15]: 0

# also fails from command line:
In [18]: ! caput otzSIM1:HDF1:Capture 1; caget otzSIM1:HDF1:Capture
Old : otzSIM1:HDF1:Capture           Done
New : otzSIM1:HDF1:Capture           Capture
otzSIM1:HDF1:Capture           Done
```

It does not work from the GUI either.  I believe the only resolution is to identify this situation and inform the user that the detector must first run a *warmup* by acquiring an image through the desired plugin chain.

prjemian, 2018-08-24:
Very likely this will happen with other area detector plugins as well.  Summary is that plugin chain needs to be *primed* with an image after certain fundamental changes such as IOC reboot, image size, binning, or bit depth, perhaps others as well.  Such logic is well beyond the scope of this issue.  Use of a diagnostic function (such as [`ad_warmed_up()`](https://github.com/NSLS-II/ophyd/issues/598#issuecomment-415201391)) is helpful at times.

prjemian, 2018-08-24:
Since there is no action recommended for ophyd, closing this issue.  Thanks all for the discussion!

danielballan, 2018-08-24:
@prjemian Are you still thinking that adding `ad_warned_up` and calling it on `PluginBase.stage()` makes sense?

prjemian, 2018-08-25:
In `stage()`, since the detector is already being configured for use, the additional test is not out of order.  Shouldn't it be part of staging only when the HDF5 file writer plugin is to be used?

danielballan, 2018-09-11:
> Shouldn't it be part of staging only when the HDF5 file writer plugin is to be used?

Yes, I think we should just add the content of `ad_warmed_up` into the `HDF5Plugin.stage()` method and raise an informative error if the `verdict` is `False`. AFAIK we are still undecided on whether other file plugins likewise require a warm-up or if it's just HDF5, In any case, we can add code to their `stage()` methods too if we need to.

prjemian, 2018-09-11:
I'm certain other file plugins will need to be *warmed up* (perhaps *primed* as more apropo).  Conditions for this are whenever the image size changes (bits, pixels) or the IOC is restarted.

Mark says that the HDF5 file writer has a flag to defer opening the HDF5 file until the time when the frame is actually to be written.  We can avoid the warmed_up check with this but the risk is that the file open (and write the various HDF structure) process may take long enough that it impacts the data acquisition rate or drops the next frame.  That depends on many other factors.  Your mileage will vary.

danielballan, 2018-09-11:
> We can avoid the warmed_up check with this but the risk is that the file open (and write the various HDF structure) process may take long enough that it impacts the data acquisition rate or drops the next frame. That depends on many other factors. Your mileage will vary.

We usually err on the side of "fail early". I think staging is a good time to do this check. I think we're on the same page then; let me know if not. :- )

prjemian, 2018-09-11:
I agree with you

tacaswell, 2019-03-11:
(re) closed by https://github.com/NSLS-II/ophyd/pull/647

closed 2019-03-11 by tacaswell