bluesky/ophyd #608: add a AD IOC versions to travis

tacaswell, 2018-09-10:
We need to do a better job of testing the universe of AD versions we actually support.

at NSLS-II we have at least (1-9-1, 2-4, 2-5, 2-6, 3-1, 3-2, 3-3, master-branch)

@teddyrendahl @prjemian Do you have a good inventory of what versions you need to support?

prjemian, 2018-09-10:
we not deploying anything older than 2-4 and have not yet deployed 3-3, we try to deploy versioned releases rather than master branch

tacaswell, 2018-09-10:
See https://github.com/NSLS-II/caproto/blob/master/.travis.yml and https://github.com/klauer/epics-on-travis for a better pattern of how to do many epics versions on travis (rather than the docker based approach that ophyd is currently using).

teddyrendahl, 2018-09-10:
We are not using any 3.3 yet. Largely 2.1 and 2.6 on the floor.