bluesky/ophyd #624: Notice when signals enter alarm state

tacaswell, 2018-10-19:
See https://github.com/NSLS-II/bluesky/issues/1100

danielballan, 2018-10-19:
We don’t already do this? Oof.

I plan to look into this as part of leveraging caproto to produce better error messages.

tacaswell, 2018-10-19:
Motors do it, but not generic signals.

tacaswell, 2018-10-19:
we should also think about how to do alarm propagation in Devices in general.  As we combine many records into a Device we will in general have inconsistent alarm states across it's children.  In many cases if there is an alarm (of some degree) on any child we want to kill the status objects.  This probably also means we need a coherent way to keep track of where we have stashed the status object.

ambarb, 2018-12-18:
@tacaswell this is the other issue that I just spoke with you about.  You converted my Bluesky issue into an ophyd issue.