bluesky/ophyd #626: Motor Subscribes in Inititalization

teddyrendahl, 2018-10-24:
## Description
`ophyd.EpicsMotor` subscribes to both the readback and `DMOV` flag on initialization.
https://github.com/NSLS-II/ophyd/blob/011c3e4a7d60a0eb98ae8cc669208d9984fca875/ophyd/epics_motor.py#L81

## Why does this matter?
1. In general we try to avoid connecting to any signals on `init`. This means any network hiccup or IOC down prevents the device from being loaded. Also prevents offline loading as discussed (#578)
2. The callback for the `DMOV` flag has some high level logging messages. This means that even if you sit with a blank screen and someone is moving one of those motors anywhere else and error occurs it shows up in your screen. Maybe this is desirable, but it is confusing when you are running a scan and an error message from another device shows up in the middle.

https://github.com/NSLS-II/ophyd/blob/011c3e4a7d60a0eb98ae8cc669208d9984fca875/ophyd/epics_motor.py#L258

danielballan, 2018-10-24:
I suspect this is because we haven’t revisited this since we added “stage” to the ophyd lexicon. 👍 for moving from init to stage.

klauer, 2018-10-24:
I think I'd still prefer to keep the state of `EpicsMotor` up-to-date regardless of whether it's being used in a scan.

If we had connection/disconnection hooks as in #607, subscription setup could be moved there. (Similarly, areaDetector plugin checking could be moved to `on_connection` and make everyone happier.)

`stage`, indicating the motor is used in a scan, could fix up (2). When used in a scan, those log messages probably should be very prominent.

danielballan, 2018-10-24:
That seems reasonable to me.

klauer, 2018-11-16:
Closed by #630

closed 2018-11-16 by klauer