bluesky/ophyd #643: Deprecate / remove EpicsSignal pv_kw init argument

klauer, 2018-11-12:
The intent of `EpicsSignal` is to abstract away pyepic's `PV` interface. However, `pv_kw` in the EpicsSignal initializer get passed through to `epics.PV` breaks this abstraction and should be removed.

I don't think anyone is using this in the wild, but it would be wise to check.

Additionally, keyword arguments are passed through for `EpicsSignal.put` and `EpicsSignal.get`. While I suppose that ship has sailed as they are assuredly used, perhaps we can restrict the keyword arguments we support in those cases and remove `**kwargs`.

Thoughts? @danielballan @tacaswell @teddyrendahl @ZLLentz

prjemian, 2018-11-12:
not used at APS

ZLLentz, 2018-11-12:
I don't think this is used at SLAC. I thought I needed it once but I was incorrect. I agree with the intent of this issue, but we should also see if any of the features here could have use cases that we want to make explicit arguments.

tacaswell, 2018-11-13:
Verified it is not using by any profile which are committed to git.

PR #651 (2018-11-19): API: remove pv_kw from EpicsSignal classes
Closes #643 

In the issue, @ZLLentz brought up the following:
>  I agree with the intent of this issue, but we should also see if any of the features here could have use cases that we want to make explicit arguments.

The keywords in pyepics `epics.PV` are as follows:
```python
    def __init__(self, pvname, callback=None, form='time',
                 verbose=False, auto_monitor=None, count= None,
                 connection_callback=None,
                 connection_timeout=None, access_callback=None):
```
I believe we are good as-is. Thoughts?
commit 25f4d91c (2018-11-19): API: remove pv_kw from EpicsSignal classes

Closes #643
closed 2018-11-19 by tacaswell