bluesky/ophyd #644: Discuss Device hierarchy walking functions

klauer, 2018-11-13:
In #630, `ophyd.Device` gained better support for iterating through the full tree of components and signals:
    1. `Device.walk_components` (classmethod - give me all `Components` and `DynamicDeviceComponents`)
    2. `Device.walk_signals` (instancemethod - give me all `Signal`s)
    3. `Device.walk_subdevice_classes` (classmethod - give me all sub-[sub-]`Device` classes)
    4. `Device.walk_subdevices` (instancemethod - give me all sub-[sub-]`Device` instances)

There are two distinct categories of the above - for iterating through Components of non-instantiated Devices, and iterating through all signals/sub-devices of an instantiated Device.

To all who might be interested in these functions, do you agree or disagree with the implementation? 

The API should be solidified prior to the next release. (So speak now, or forever hold your peace...)

ZLLentz, 2019-01-09:
I'm a fan of these in general, but I'll reiterate that I think the return types should be consistent. Currently half of these return `ComponentWalk` `namedtuples`, while the others return simple `tuples`. I like the `namedtuples`, so that code I write that accesses these fields is very clear, e.g. perhaps for debugging purposes I have `pvnames = list(walk.item.pvname for walk in device.walk_signals())`.

These may also be useful for applications built on top of `ophyd` like SLAC's `typhon`.

tacaswell, 2019-01-25:
Do we need all the flavors?  I think we can get away with just a class method of "get me all of the Components" and an instance method of "get me everything that is instantiated".

I have previously argued that the distinction between Device and Signal should be made fuzzier, particularly as we move to v4 (or tango).

klauer, 2019-01-25:
@ZLLentz - agreed, if these functions stay in as-is, they should all return namedtuples.

@tacaswell - my opinion is that these are all useful in different scenarios. If some form of them did not exist, callers would just be reimplementing the functionality based on the walk functions that did exist. Regarding blurring the lines of Device/Signal, I think we need a much larger discussion on that (I'll open an issue soon for its discussion).

klauer, 2020-11-03:
These are baked into several releases and are only part of a near 2 year old issue. 

If you intend to change this, make a new issue and reference this one.

closed 2020-11-03 by klauer