bluesky/ophyd #663: Custom filenames for Bluesky scans

bsobhani, 2018-12-18:
When using the scan_1d function, bluesky overwrites the user-specified filename in the IOC's HDF5 plugin with the scan's uid. Is it possible to get the scan to save images using a custom filename, instead of the scan's uid?

tacaswell, 2018-12-18:
Moved this from bluesy because this is really an ophyd issue (from the bluesky side we just say "stage").  The question here is if we want to add hooks into the stage / trigger API to allow user-defined names to be directly injected.

danielballan, 2018-12-18:
> Also, I want to learn how you moved this issue. 

That's a new "beta" GitHub feature. Look in the right-side bar for "Transfer this issue." Screenshot:

[image]

tacaswell, 2018-12-18:
> The user should be advised when a filename (absolute reference) will be overwritten.

We should talk to Mark about getting this as a PV in the FilePlugins in AD.

stuwilkins, 2018-12-19:
@tacaswell 

Do you mean something like the PV flag that is given by:

`$(P)$(R)FilePathExists_RBV`

To indicate that the path exists? That would seem very easy to do.

tacaswell, 2018-12-20:
@stuwilkins Yes, exactly that :+1: .