bluesky/ophyd #667: Python-level dead band

danielballan, 2018-12-20:
Different experiments might call for different levels of precision in tracking changes in, say, the beam current. One user might want a detailed data dump, another might want a shorter table that only records larger shifts. (And all the detailed data is being captured in the archiver anyway.)

We can add a Python-level deadband in the monitor message or in the `Device.subscribe` method to enable this.

via a conversation with @cmazzoli

prjemian, 2018-12-20:
Checking the [ao record documentation](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-9.html), there is the ['MLST'](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-9.html#HEADING9-47) field (Last Value Change Monitor Trigger Value) that is *almost* what issue describes.  Changing this value changes when the PV monitors are triggered.  Close, but prolly not close enough.  We need a python-specific value?

prjemian, 2018-12-20:
Oh, nevermind since MLST can't be modified from a CA client.  At least we looked.

danielballan, 2020-09-01:
There is renewed interest in this, prompted by @EliotGann.