bluesky/ophyd #668: v1.3.1 bug fix release

tacaswell, 2019-01-03:
We would like to do a bug-fix release for ophyd to support the upcoming NSLS-II cycle which is more conservative.

I have pushed a branch with all of these changes to upstream to get CI to run on it, see https://github.com/NSLS-II/ophyd/compare/v1.3.x?expand=1 for the differences.

# PRs to backport

 - #609 yes TBD
 - #661 yes 438e2391eda0fefc03378bb6186281a590319f5f
 - #649 yes 1b340b52f410e08af5c69710f7234ae466332417
 - #629 yes 7966c118afce4af0b5fcb849cd031b50849f913d
 - #611 yes 8c96021405ce4a4145f208f2a69d4bfae28e8e06
 - #633 yes d7c4cca8bfd1b830869cc112153d3b6497f2ba67
 - #634 yes dd6a3a0335c7e9d6fe6fc0beb121977cfe637aa7
 - #627 yes 7ed44a84e7c6762277be5d7b445a43946f13b6a0
 - #610 yes d2579759559bc7074ddd4f9eff7c4d80b217134c
 - #589 yes 011c3e4a7d60a0eb98ae8cc669208d9984fca875
 - #612 yes b907a8c967a8eb4353666b8dd632d87bd1716e2c
 - #595 yes 112c88968e1965803868195818d4ae122b742ee0
 - #606 yes 3ef7f558c90febe890d65e1de5de9323bb09fbd3

# PRs not to backport 

 - #650 no
 - #655 no
 - #654 no
 - #623 no
 - #651 no
 - #646 no
 - #648 no
 - #583 no
 - #645 no
 - #642 no
 - #641 no
 - #630 no
 - #631 no
 - #636 no
 - #625 no
 - #604 no

danielballan, 2019-01-03:
Looks fine to me.

danielballan, 2019-01-03:
Actually you missed #664 entirely. It's not a bug fix, but it doesn't touch any existing code, and it would make the logging procedure uniform across caproto, ophyd, bluesky.

mrakitin, 2019-01-03:
+1 for releasing it.

tacaswell, 2019-01-03:
leaving this open as a reminder to merge the 1.3.x branch into master.  The plan is to `git merge -s ours v1.3.x` to link the commits (so the tag can be reached from master which make `git describe` give the right answer) but will not pull any changes from the 1.3.x branch into master.