bluesky/ophyd #673: derived signal can't handle type change

licode, 2019-01-11:


tacaswell, 2019-01-14:
@klauer Does #653 also fix this?

klauer, 2019-01-14:
@tacaswell It's not entirely clear to me what this issue means

tacaswell, 2019-01-14:
at LIX they are doing string -> float conversion for vacuum pressures:

```python
class _MaxiPressure(DerivedSignal):
    def inverse(self, value):
        msg = value
        if msg == "UNDER":
            return 1.0e-3
        elif msg == "OVER":
            return 1.1e3
        else:
            return float(msg)

    def describe(self):
        desc = super().describe()
        desc[self.name]['dtype'] = 'float'
        return desc

    def put(self, value, **kwrags):
        raise Exception("gauge is read only")

```

PR #677 (2019-01-26): ENH: NDDerivedSignal
# Description
This introduces `NDDerivedSignal` a subclass of `DerivedSignal` to reshape a flat array into a shaped one based on either static values or other components of the class. This was made with the purpose of the supporting shaped image signals, but I believe it is general enough to apply to any situation where you have a flat array this is intended to be reshaped.

# Motivation
This is useful for a variety of reasons, my major motivation is for `typhon` which can not tell a flat array from an image without checking if we are looking at an `AreaDetector` class exp [... cut]
closed 2019-01-26 by klauer