bluesky/ophyd #683: Fix tab-completion of subdevices.

danielballan, 2019-02-06:
`device.<TAB>` used to show _all_ the children. I think the refactoring around `__getattr__` broke this (but I'm working from memory). Anyway, should be easy to fix using the new walking methods.

tacaswell, 2019-02-06:
To clarify a bit what I have seen:

`dev.<TAB>` gets the first generation but `dev.child.<TAB>` gets nothing.

klauer, 2019-03-04:
I can't seem to reproduce on the current master 05f3770:

[image]

mrakitin, 2019-03-04:
@klauer, on our side it was fixed by making sure we don't use the `jedi` package, i.e.:
```
ipython --IPCompleter.use_jedi=False
```
See the context in https://github.com/ipython/ipython/issues/11546. Which version of IPython do you have? Do you have `jedi` installed?

klauer, 2019-03-04:
@mrakitin - interesting. I had never observed the issue, personally.

Should this be closed, then? Or should there be a documentation note regarding jedi?

mrakitin, 2019-03-04:
I'm fine with closing this issue. I like the idea of having a note in the Ophyd documentation about a possible issue with jedi + ipython 7.2.x.

danielballan, 2019-03-05:
Should we ban IPython 7.2.x in the tutorial and in our conda metapackages?

prjemian, 2019-04-15:
So what is the recommendation here? We have ipython 7.2.0 and jedi 0.13.2.  This keeps biting us.

mrakitin, 2019-04-15:
At NSLS-II we are still using the workaround explained in https://github.com/NSLS-II/ophyd/issues/683#issuecomment-469388963.