bluesky/ophyd #713: staging a EpicsSignalWithRBV leads to a timeout error

KedoKudo, 2019-04-24:
When adding an EpicsSignalWithRBV (for example, acquire_period in det.cam) to the staging dictionary, the __set_and_wait()__ called by stage() will be stuck in the waiting loop, resulting in a timeout error that breaks the scan.  

However, if the PV is set directly using the __put()__ function (det.cam.acquire_perido.put(1)), the PV can be updated without any issue.

danielballan, 2019-04-24:
The `set_and_wait` is designed to block until the value is updated. Using `put` will _attempt_ to update the value but will not ensure that it is done before the staging continues.

KedoKudo, 2019-04-24:
Does this mean that an instance of EpicsSignalWithRBV should only be updated using put() instead of using stage()?

danielballan, 2019-04-24:
I would look into why `set_and_wait` is getting stuck. It sounds like the PV that it is setting is never updating to the desired value. That could be an issue on the IOC side. I would start by monitoring that PV in a separate terminal while staging runs.

tacaswell, 2019-04-24:
Also remember that with AD you are putting to one PV and getting from another.

We are currently having a bunch of issues like this at NSLS-II, what detector and AD version are you having this issue with?

attn @kgofron

KedoKudo, 2019-04-24:
We are using a pointgray detector and the AD version should be 2.7.   @cpchuang knows more about the hardware details.

tacaswell, 2019-04-24:
Is this an all the time thing or a every-so-often thing?

cpchuang, 2019-04-24:
> We are using a pointgray detector and the AD version should be 2.7. @cpchuang knows more about the hardware details.

Model: Grasshopper3 GS3-U3-23S6M
Firmware: 2.30.3.0
AD Vers: 2.9.3

tacaswell, 2019-04-26:
@kgofron will be at APS next week, hopefully this can be investigated in person.

prjemian, 2019-04-26:
We'll do our best to capture that discussion here
Others at APS such as @decarlof & @jilavsky might be interested

prjemian, 2019-04-30:
This is related to #598 : the area detector plugin has not yet been primed (warmed up, had an NDarray pushed to it, ...).  Here is the message on the area detector's IOC console as the detector is staged:

```
epics> 2019/04/30 11:09:44.034 NDPluginFile::doCapture: ERROR, must collect an array to get dimensions first
```

This is with the SimDetector, the HDF5 file writer plugin and AD3.2, EPICS 7.0.2
I do not believe this is specific to any specific detector nor any ophyd version.

The timeout is reported on the `Capture` signal with this message:

```
TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='vm7SIM1:HDF1:Capture_RBV', name='adsimdet_hdf1_capture', parent='adsimdet_hdf1', value=0, timestamp=1556639274.439258, pv_kw={}, auto_monitor=False, string=False, write_pv='vm7SIM1:HDF1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.
```

Write to the `Capture` signal is part of the default HDF plugin staging:

```
In [13]: adsimdet.hdf1.stage_sigs
Out[13]: 
OrderedDict([('enable', 1),
             ('auto_increment', 'Yes'),
             ('array_counter', 0),
             ('auto_save', 'Yes'),
             ('num_capture', 0),
             ('file_template', '%s%s_%6.6d.h5'),
             ('file_write_mode', 'Stream'),
             ('capture', 1),
             ('blocking_callbacks', 'Yes'),
             ('parent.cam.array_callbacks', 1)])
```

Note these versions:

```
RE.md['APSTOOLS_VERSION'] = 1.1.0
RE.md['BLUESKY_VERSION'] = 1.5.2
RE.md['OPHYD_VERSION'] = 1.3.2
```

prjemian, 2019-04-30:
Ophyd is still reporting a TimeoutError when it should be reporting a misconfiguration error that AD needs to push an image to the plugin before ophyd can stage the detector.  To report this error, it needs to test and this test is missing from the code base now.

The test we should implement somewhere (generally) in `ophyd.areadetector` applies checks for the timestamp and array size of the image buffer in the plugin (whether that is TIFF, HDF5, or other writer).  For example:

```
In [16]: adsimdet.hdf1.time_stamp.value
Out[16]: 0.0
In [19]: adsimdet.hdf1.array_size.height.value
Out[19]: 0

In [20]: adsimdet.hdf1.array_size.width.value
Out[20]: 0

In [21]: adsimdet.hdf1.array_size.depth.value
Out[21]: 0
```
If these are all zero (as shown here), then the plugin needs to be primed and ophyd should raise an Exception with the same text that AD reports on the IOC console:

```
ERROR, must collect an array to get dimensions first
```

klauer, 2019-04-30:
Can you reproduce this with the out-of-the-box SimDetector, @prjemian? I'd like to investigate, but would need to duplicate your test setup.

tacaswell, 2019-04-30:
Does that failure set any alarm status on the PV?

I could have sworn we did have a check that `prod(self.array_size.get() != 0` someplace but apparently not?

prjemian, 2019-04-30:
This was the out of the box sim detector.  Compiled in February which is recent but still the master has moved along.  I believe I would get the same result by pulling master and rebuilding.  Probably take me two hours of undistracted time though.

tacaswell, 2019-04-30:
found it: https://github.com/NSLS-II/ophyd/pull/647

prjemian, 2019-04-30:
Yes.  That first message confuses the daylights out of me.  What does it say there?

prjemian, 2019-04-30:
But yes, exactly, plus a check on the time stamp which will be greater than zero for *any* image array.

prjemian, 2019-04-30:
Not seeing any code that checks array_size directly: https://github.com/NSLS-II/ophyd/search?q=array_size&unscoped_q=array_size

Looks like Ken's fix_plugin_primed code branch for PR #598 was deleted.  This might have had the fix.

prjemian, 2019-04-30:
It would be nice if area detector made a PV that is set when this situation occurs.  Since it does not, at least we need a test as described above.

prjemian, 2019-04-30:
@KedoKudo: Can you report the full exception trace you receive?  We may be solving a different valuable problem.

tacaswell, 2019-04-30:
It seems to be on master, but not being found by the search (?!)

https://github.com/NSLS-II/ophyd/blob/cf2b85c2479d645c144289fca79f74b151183417/ophyd/areadetector/plugins.py#L917-L922

tacaswell, 2019-04-30:
We have also had issues with changing the ROI size and then saving where the plugin is warmed up, but is expecting a different size than it gets and errors out in weird was when saving the frame.

Probably we should be walking the plugin chains and asserting that everything agrees that the array sizes match.

prjemian, 2019-04-30:
Yes, Mark Rivers commented that is another reason to prime the AD plugin chain.  However, in that case, the tests above do not inform us.  Also, the IOC console reports a separate problem that `Capture` cannot be set in `Single` image mode.

tacaswell, 2019-04-30:
> Does that failure set any alarm status on the PV?

By this I mean, if we try to set a PV to a value that the IOC thinks is invalid, does it set any alarm state we can watch for or provide any feedback on the `WriteNotify`?

prjemian, 2019-04-30:
In this case, I do not see any PV on my screens.

prjemian, 2019-04-30:
Looking at my local installation of ophyd 1.3.2 in `bluesky/lib/python3.6/site-packages/ophyd/areadetector/plugins.py` in `class HDF5Plugin(FilePlugin):`, there is no `stage()` method.  Here's what I've got (starts on line 820):

```
class HDF5Plugin(FilePlugin):
    _default_suffix = 'HDF1:'
    _suffix_re = 'HDF\d:'
    _html_docs = ['NDFileHDF5.html']
    _plugin_type = 'NDFileHDF5'

    _default_configuration_attrs = (FilePlugin._default_configuration_attrs + (
        'boundary_align',
        'boundary_threshold',
        'compression',
        'data_bits_offset',
        'extra_dim_name',
        'extra_dim_size',
        'io_speed',
        'num_col_chunks',
        'num_data_bits',
        'num_extra_dims',
        'num_frames_chunks',
        'num_frames_flush',
        'num_row_chunks',
        'run_time',
        'store_attr',
        'store_perform',
        'szip_num_pixels',
        'zlevel')
    )

    boundary_align = C(SignalWithRBV, 'BoundaryAlign')
    boundary_threshold = C(SignalWithRBV, 'BoundaryThreshold')
    compression = C(SignalWithRBV, 'Compression')
    data_bits_offset = C(SignalWithRBV, 'DataBitsOffset')

    extra_dim_name = DDC(ad_group(EpicsSignalRO,
                                  (('name_x', 'ExtraDimNameX_RBV'),
                                   ('name_y', 'ExtraDimNameY_RBV'),
                                   ('name_n', 'ExtraDimNameN_RBV'))),
                         doc='Extra dimension names (XYN)',
                         default_read_attrs=('name_x', 'name_y', 'name_n'))

    extra_dim_size = DDC(ad_group(SignalWithRBV,
                                  (('size_x', 'ExtraDimSizeX'),
                                   ('size_y', 'ExtraDimSizeY'),
                                   ('size_n', 'ExtraDimSizeN'))),
                         doc='Extra dimension sizes (XYN)',
                         default_read_attrs=('size_x', 'size_y', 'size_n'))

    io_speed = C(EpicsSignal, 'IOSpeed')
    num_col_chunks = C(SignalWithRBV, 'NumColChunks')
    num_data_bits = C(SignalWithRBV, 'NumDataBits')
    num_extra_dims = C(SignalWithRBV, 'NumExtraDims')
    num_frames_chunks = C(SignalWithRBV, 'NumFramesChunks')
    num_frames_flush = C(SignalWithRBV, 'NumFramesFlush')
    num_row_chunks = C(SignalWithRBV, 'NumRowChunks')
    run_time = C(EpicsSignal, 'RunTime')
    szip_num_pixels = C(SignalWithRBV, 'SZipNumPixels')
    store_attr = C(SignalWithRBV, 'StoreAttr')
    store_perform = C(SignalWithRBV, 'StorePerform')
    zlevel = C(SignalWithRBV, 'ZLevel')

    def warmup(self):
        """
        A convenience method for 'priming' the plugin.

        The plugin has to 'see' one acquisition before it is ready to capture.
        This sets the array size, etc.
        """
        set_and_wait(self.enable, 1)
        sigs = OrderedDict([(self.parent.cam.array_callbacks, 1),
                            (self.parent.cam.image_mode, 'Single'),
                            (self.parent.cam.trigger_mode, 'Internal'),
                            # just in case tha acquisition time is set very long...
                            (self.parent.cam.acquire_time, 1),
                            (self.parent.cam.acquire_period, 1),
                            (self.parent.cam.acquire, 1)])

        original_vals = {sig: sig.get() for sig in sigs}

        for sig, val in sigs.items():
            ttime.sleep(0.1)  # abundance of caution
            set_and_wait(sig, val)

        ttime.sleep(2)  # wait for acquisition

        for sig, val in reversed(list(original_vals.items())):
            ttime.sleep(0.1)
            set_and_wait(sig, val)
```

So, master has code that is not in 1.3.2.  Can we add this to the [cherry picking festival](https://github.com/NSLS-II/ophyd/issues/714#issuecomment-488016308) for today?

klauer, 2019-05-09:
I think it's probably fair to say that the original cause of the timeout noted in this issue was due to the unprimed HDF5 plugin, which will at least raise an exception in a later (as of yet, unreleased) version.

If that's the case, I think we can safely close this.

prjemian, 2019-05-10:
Agreed

cpchuang, 2019-05-10:
Did some test today, still get timeout error from "AcquirePeriod_RBV", log below...

```python
In [20]: cam.stage_sigs                                                                                                                                                      
Out[20]: 
OrderedDict([('num_images', 1),
             ('image_mode', 'Multiple'),
             ('trigger_mode', 'Internal'),
             ('acquire_time', 0.05),
             ('gain', 5),
             ('acquire_period', 0.06)])

In [21]: det.stage()                                                                                                                                                         
---------------------------------------------------------------------------
TimeoutError                              Traceback (most recent call last)
<ipython-input-21-5bf3da16868d> in <module>
----> 1 det.stage()

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/areadetector/trigger_mixins.py in stage(self)
    119     def stage(self):
    120         self._acquisition_signal.subscribe(self._acquire_changed)
--> 121         super().stage()
    122 
    123     def unstage(self):

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/areadetector/base.py in stage(self, *args, **kwargs)
    156 
    157     def stage(self, *args, **kwargs):
--> 158         ret = super().stage(*args, **kwargs)
    159         try:
    160             self.validate_asyn_ports()

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/device.py in stage(self)
    605                 device = getattr(self, attr)
    606                 if hasattr(device, 'stage'):
--> 607                     device.stage()
    608                     devices_staged.append(device)
    609         except Exception:

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/areadetector/base.py in stage(self, *args, **kwargs)
    156 
    157     def stage(self, *args, **kwargs):
--> 158         ret = super().stage(*args, **kwargs)
    159         try:
    160             self.validate_asyn_ports()

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/device.py in stage(self)
    596                                self.name,
    597                                val, original_vals[sig])
--> 598                 set_and_wait(sig, val)
    599                 # It worked -- now add it to this list of sigs to unstage.
    600                 self._original_vals[sig] = original_vals[sig]

~/opt/miniconda3/envs/bluesky_py36/lib/python3.6/site-packages/ophyd/utils/epics_pvs.py in set_and_wait(signal, val, poll_time, timeout, rtol, atol)
    263             raise TimeoutError("Attempted to set %r to value %r and timed "
    264                                "out after %r seconds. Current value is %r." %
--> 265                                (signal, val, timeout, current_value))
    266 
    267 

TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='1idPG3:cam1:AcquirePeriod_RBV', name='det_cam_acquire_period', parent='det_cam', value=0.0061181641399925876, timestamp=1557507170.986168, tolerance=0.001, pv_kw={}, auto_monitor=False, string=False, write_pv='1idPG3:cam1:AcquirePeriod', limits=False, put_complete=False) to value 0.06 and timed out after 10 seconds. Current value is 0.0061181641399925876.
```

cpchuang, 2019-05-10:
"1idPG3:cam1:FrameRateAutoMode_RBV" has to switch to "manual" before staging. Otherwise it leads to "AcquirePeriod" timeout error.

KedoKudo, 2019-05-10:
> "1idPG3:cam1:FrameRateAutoMode" has to switch to "manual" before staging. Otherwise, it leads to "AcquirePeriod" timeout error.

Just a quick follow up on @cpchuang's comment.  It seems like the time-out error is related to the IOC design where the auto_mode will prevent BlueSky&Ophyd from updating the acquire_period value.  The affected PVs we found so far include:

| Affected PV  | Solution |
| ------------: |:---------| 
|cam.frame_rate | cam.frame_rate_auto_mode.put(0) |
|cam.gain |  cam.gain_auto_mode.put(0) |
|cam.acquire_period | cam.acquire_period_auto_mode.put(0) |
 
Since this is related to the design of the area detector control system, I think we are going to do an initialization at the beginning of each experiment to turn off all related auto_mode.

closed 2019-05-10 by KedoKudo

klauer, 2019-05-10:
Thanks for the follow-up on the details of this, @cpchuang and @KedoKudo! AreaDetector configuration can be rather complicated at times.

We might want to bring in that staging logic to ophyd when we expand on our cam support (much as we recently did with extended plugin version support in #674).