bluesky/ophyd #714: TypeError in ScalerCH.select_channels(None)

prjemian, 2019-04-30:
Seems that the [code committed](https://github.com/NSLS-II/ophyd/commit/543e7ef81f3cb760192a0de719e51f9359642ae8) as part of #692 did not make it into tag 1.3.2, thus my local code expecting this fix fails like this:

```
~/.ipython/profile_bluesky/startup/20-detectors.py in <module>()
      4 
      5 scaler = ScalerCH('vm7:scaler1', name='scaler')
----> 6 scaler.select_channels(None)
      7 

~/Apps/anaconda/envs/bluesky/lib/python3.6/site-packages/ophyd/scaler.py in select_channels(self, chan_names)
    136 
    137         read_attrs = ['chan01']  # always include time
--> 138         for ch in chan_names:
    139             try:
    140                 read_attrs.append(name_map[ch])

TypeError: 'NoneType' object is not iterable
```

The fix *does exist* in the master branch now.  Can a new tag be made in the next week?

prjemian, 2019-04-30:
Example [local workaround](https://github.com/prjemian/ipython-vm7/commit/57f4e4365df58869f5025acd8759e9b66ce6cd08) until the new tag.

danielballan, 2019-04-30:
Thanks for sharing the workaround. We could at least tag a beta this week. @klauer Are there still known problems with signals / callback-processing in `master` that should give us pause before tagging a new stable release?

klauer, 2019-04-30:
Yes, this needs resolving: https://github.com/NSLS-II/ophyd/issues/708

We're looking to a new pyepics release soon, but there is not yet a fix designed for this issue. It likely will cause issues with PV reusage during a session. As far as I am aware, the refactor has not been extensively tested outside of CI.

tacaswell, 2019-04-30:
Ah, when I did the backports I looked at merged PRs, it looks like that commit got pushed directly to master so I missed it.

As @klauer points out master currently needs a bit more work before it can be tagged (and has some pretty big improvements (but given the rules of un-intended consequences means will also break something :disappointed: ) ).

Unless someone protests I'll cherry-pick that commit back to 1.3.x and tag 1.3.3 this afternoon?

mrakitin, 2019-04-30:
That's a good plan, @tacaswell.

danielballan, 2019-05-02:
Done and released on PyPI. Conda packages in progress.

closed 2019-05-02 by danielballan

prjemian, 2019-05-02:
Thanks!  Conda package coming or I switch to pip on this as proof-of-fix?

prjemian, 2019-05-02:
I'm blind.

mrakitin, 2019-05-02:
New conda packages are available at https://anaconda.org/lightsource2-tag/ophyd/files.