bluesky/ophyd #717: networkx has no attribute 'utils'

gwbischof, 2019-05-01:
handler_registry:                                                         
        NPY_SEQ: ophyd.sim.NumpySeqHandler 

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-28-8bc71255a22e> in <module>
----> 1 a[1]

~/conda_envs/garrett/lib/python3.7/site-packages/intake/catalog/entry.py in __getitem__(self, item)
    145             else:
    146                 item = item[0]
--> 147         return self._get_default_source()[item]
    148 
    149     def __repr__(self):

~/conda_envs/garrett/lib/python3.7/site-packages/intake/catalog/entry.py in _get_default_source(self)
    112     def _get_default_source(self):
    113         """Instantiate DataSource with default agruments"""
--> 114         return self()
    115 
    116     @property

~/conda_envs/garrett/lib/python3.7/site-packages/intake/catalog/entry.py in __call__(self, persist, **kwargs)
     98             raise ValueError('Persist value (%s) not understood' % persist)
     99         persist = persist or self._pmode
--> 100         s = self.get(**kwargs)
    101         if s.has_been_persisted and persist is not 'never':
    102             s2 = s.get_persisted()

~/conda_envs/garrett/lib/python3.7/site-packages/intake/catalog/local.py in get(self, **user_parameters)
    272         """Instantiate the DataSource for the given parameters"""
    273         plugin, open_args = self._create_open_args(user_parameters)
--> 274         data_source = plugin(**open_args)
    275         data_source.catalog_object = self._catalog
    276         data_source.name = self.name

~/dama/intake-bluesky/intake_bluesky/mongo_normalized.py in __init__(self, metadatastore_db, asset_registry_db, handler_registry, query, **kwargs)
     61         self._query = query or {}
     62         if handler_registry is None:
---> 63             handler_registry = {}
     64         parsed_handler_registry = parse_handler_registry(handler_registry)
     65         self.filler = event_model.Filler(parsed_handler_registry)

~/dama/intake-bluesky/intake_bluesky/core.py in parse_handler_registry(handler_registry)
    680         if isinstance(handler_str, str):
    681             module_name, _, class_name = handler_str.rpartition('.')
--> 682             class_ = getattr(importlib.import_module(module_name), class_name)
    683         else:
    684             class_ = handler_str

~/conda_envs/garrett/lib/python3.7/importlib/__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _gcd_import(name, package, level)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _find_and_load(name, import_)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _gcd_import(name, package, level)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _find_and_load(name, import_)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _load_unlocked(spec)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap_external.py in exec_module(self, module)

~/conda_envs/garrett/lib/python3.7/importlib/_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~/conda_envs/garrett/lib/python3.7/site-packages/ophyd/__init__.py in <module>
     87                      ALL_COMPONENTS, kind_context)
     88 from .status import StatusBase
---> 89 from .mca import EpicsMCA, EpicsDXP
     90 from .quadem import QuadEM, NSLS_EM, TetrAMM, APS_EM
     91 

~/conda_envs/garrett/lib/python3.7/site-packages/ophyd/mca.py in <module>
      8 from .device import (Device, Component as C, DynamicDeviceComponent as DDC,
      9                      Staged, BlueskyInterface, ALL_COMPONENTS, Kind)
---> 10 from .areadetector import EpicsSignalWithRBV as SignalWithRBV
     11 
     12 

~/conda_envs/garrett/lib/python3.7/site-packages/ophyd/areadetector/__init__.py in <module>
      6 logger.addHandler(logging.NullHandler())
      7 
----> 8 from .base import *
      9 from .cam import *
     10 from .detectors import *

~/conda_envs/garrett/lib/python3.7/site-packages/ophyd/areadetector/base.py in <module>
      4 import sys
      5 from collections import OrderedDict
----> 6 import networkx as nx
      7 
      8 from ..signal import EpicsSignal

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/__init__.py in <module>
    112 from networkx.relabel import *
    113 
--> 114 import networkx.generators
    115 from networkx.generators import *
    116 

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/generators/__init__.py in <module>
     12 from networkx.generators.expanders import *
     13 from networkx.generators.geometric import *
---> 14 from networkx.generators.intersection import *
     15 from networkx.generators.joint_degree_seq import *
     16 from networkx.generators.lattice import *

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/generators/intersection.py in <module>
     11 import random
     12 import networkx as nx
---> 13 from networkx.algorithms import bipartite
     14 from networkx.utils import py_random_state
     15 

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/algorithms/__init__.py in <module>
     71 import networkx.algorithms.tournament
     72 import networkx.algorithms.traversal
---> 73 import networkx.algorithms.tree
     74 
     75 # Make certain functions from some of the previous subpackages available

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/algorithms/tree/__init__.py in <module>
----> 1 from .branchings import *
      2 from .coding import *
      3 from .mst import *
      4 from .recognition import *
      5 from .operations import *

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/algorithms/tree/branchings.py in <module>
     39 from networkx.utils import py_random_state
     40 
---> 41 from .recognition import *
     42 
     43 __all__ = [

~/conda_envs/garrett/lib/python3.7/site-packages/networkx/algorithms/tree/recognition.py in <module>
     86 
     87 
---> 88 @nx.utils.not_implemented_for('undirected')
     89 def is_arborescence(G):
     90     """

AttributeError: module 'networkx' has no attribute 'utils'

gwbischof, 2019-05-01:
my conda env: 

# packages in environment at /home/gbischof/conda_envs/bb:
#
# Name                    Version                   Build  Channel
anaconda-client           1.7.2                    py37_0    http://alexandria/conda/defaults
appdirs                   1.4.3                     <pip>
asciitree                 0.3.3                     <pip>
asn1crypto                0.24.0                   py37_0    http://alexandria/conda/defaults
backcall                  0.1.0                     <pip>
bluesky                   1.5.2                     <pip>
bluesky-browser           0.post100+ge6b8ab7           <pip>
ca-certificates           2018.03.07                    0    http://alexandria/conda/defaults
certifi                   2018.10.15               py37_0    http://alexandria/conda/defaults
cffi                      1.11.5           py37he75722e_1    http://alexandria/conda/defaults
cftime                    1.0.3.4                   <pip>
chardet                   3.0.4                    py37_1    http://alexandria/conda/defaults
cloudpickle               0.8.1                     <pip>
clyent                    1.2.2                    py37_1    http://alexandria/conda/defaults
cryptography              2.4.1            py37h1ba5d50_0    http://alexandria/conda/defaults
cycler                    0.10.0                    <pip>
dask                      1.2.1                     <pip>
decorator                 4.3.0                    py37_0    http://alexandria/conda/defaults
event-model               1.9.0a3                   <pip>
fasteners                 0.14.1                    <pip>
historydict               1.2.3                     <pip>
holoviews                 1.12.2                    <pip>
idna                      2.7                      py37_0    http://alexandria/conda/defaults
intake                    0.4.4                     <pip>
intake-bluesky            0.1.0a5.post1+g2bf76e8           <pip>
intake-xarray             0.3.0                     <pip>
ipython                   7.5.0                     <pip>
ipython_genutils          0.2.0                    py37_0    http://alexandria/conda/defaults
jedi                      0.13.3                    <pip>
Jinja2                    2.10.1                    <pip>
jsonschema                2.6.0                    py37_0    http://alexandria/conda/defaults
jupyter_core              4.4.0                    py37_0    http://alexandria/conda/defaults
kiwisolver                1.1.0                     <pip>
libedit                   3.1.20170329         h6b74fdf_2    http://alexandria/conda/defaults
libffi                    3.2.1                hd88cf55_4    http://alexandria/conda/defaults
libgcc-ng                 8.2.0                hdf63c60_1    http://alexandria/conda/defaults
libstdcxx-ng              8.2.0                hdf63c60_1    http://alexandria/conda/defaults
locket                    0.2.0                     <pip>
MarkupSafe                1.1.1                     <pip>
matplotlib                3.0.3                     <pip>
mongoquery                1.3.5                     <pip>
monotonic                 1.5                       <pip>
msgpack                   0.6.1                     <pip>
msgpack-numpy             0.4.4.2                   <pip>
msgpack-python            0.5.6                     <pip>
nbformat                  4.4.0                    py37_0    http://alexandria/conda/defaults
ncurses                   6.1                  he6710b0_1    http://alexandria/conda/defaults
netCDF4                   1.5.1                     <pip>
networkx                  2.3                       <pip>
numcodecs                 0.6.3                     <pip>
numpy                     1.16.3                    <pip>
openssl                   1.1.1                h7b6447c_0    http://alexandria/conda/defaults
ophyd                     1.3.2                     <pip>
pandas                    0.24.2                    <pip>
param                     1.9.0                     <pip>
parso                     0.4.0                     <pip>
partd                     0.3.10                    <pip>
pexpect                   4.7.0                     <pip>
pickleshare               0.7.5                     <pip>
pip                       18.1                     py37_0    http://alexandria/conda/defaults
prompt-toolkit            2.0.9                     <pip>
ptyprocess                0.6.0                     <pip>
pycparser                 2.19                     py37_0    http://alexandria/conda/defaults
pyepics                   3.3.3                     <pip>
Pygments                  2.3.1                     <pip>
[... 10 lines cut ...]
pyviz-comms               0.7.2                     <pip>
pyyaml                    3.13             py37h14c3975_0    http://alexandria/conda/defaults
pyzmq                     16.0.4                    <pip>
QtPy                      1.7.0                     <pip>
readline                  7.0                  h7b6447c_5    http://alexandria/conda/defaults
requests                  2.20.1                   py37_0    http://alexandria/conda/defaults
setuptools                40.6.2                   py37_0    http://alexandria/conda/defaults
six                       1.11.0                   py37_1    http://alexandria/conda/defaults
sqlite                    3.25.3               h7b6447c_0    http://alexandria/conda/defaults
[... cut ...]
wcwidth                   0.1.7                     <pip>
wheel                     0.32.3                   py37_0    http://alexandria/conda/defaults
xarray                    0.12.1                    <pip>
xz                        5.2.4                h14c3975_4    http://alexandria/conda/defaults
yaml                      0.1.7                had09818_2    http://alexandria/conda/defaults
zarr                      2.3.1                     <pip>
zlib                      1.2.11               h7b6447c_3    http://alexandria/conda/defaults

klauer, 2019-05-01:
The only line in ophyd is `import networkx as nx` - what happens when you do that in your environment?

gwbischof, 2019-05-01:
same error, I have networkx 2.3 installed

gwbischof, 2019-05-01:
hmm same error with 2.2

klauer, 2019-05-01:
OK, in that case it's not really an ophyd issue, but we should still see how to resolve it...

Brief search shows potentially related issues: https://stackoverflow.com/questions/49755099/attributeerror-module-networkx-has-no-attribute-utils
https://stackoverflow.com/questions/42084985/attributeerror-module-networkx-has-no-attribute-graph

gwbischof, 2019-05-01:
I switched back and forth a between 2.2 and 2.3, now it is working with both 2.2 and 2.3. I also reset the kernels when switching versions. I tested with ipython and jupyter notebook.  Everything is good now. 
Maybe the issue was like they stated in your first stackoverflow link: https://stackoverflow.com/a/49811382/6513183

closed 2019-05-01 by gwbischof

klauer, 2019-05-01:
👍 Great! Glad to hear it was resolved, @gwbischof.

gwbischof, 2019-05-01:
Thanks for the help