bluesky/ophyd #728: TimeoutError when setting areaDetector file path

prjemian, 2019-05-24:
When using `yield from bps.mv(areadetector.tiff1.file_path, "/tmp/one/two/three/four/five")`, a TimeoutError is received (eventually).  Note that `abs_set` returns with no error.  Same behavior is observed for the HDF5 plugin.  Assume it's a common problem for all file writer plugins.

The problem is that `$(P):TIFF1:CreateDirectory` is set to 0 (default setting) so the AD TIFF plugin does not create the additional directory paths.

First off, this should not cause a TimeoutError since AD has a PV that describes if setting the file path succeeded (`$(P):TIFF1:FilePathExists_RBV`).  This may need custom handling in ophyd `.set()` or `.put()` (which one?).

Second, the user should take control of this PV (needs an interface in ophyd) and set it to `-len(new_file_path.split("/"))` (negative number and then number of subdirectories to be created).

prjemian, 2019-05-24:
How to avoid the TimeoutError by checking the status PV?  Any takers?

tacaswell, 2019-05-24:
I would not expect `abs_set` to fail as it does not wait for the signal to report if it actually succeeded in setting itsself.  

In the case of `EpicsSignalWithRBV` a signal reports that it is sucessfully set when the value in the  `*_RBV` is equal to the set point.  The reason you are getting a timeout is that AD adds a trailing slash to the filepath so the `_RBV` will never be equal to the set point.   We deal with this in the filestore (should be re-named assert registry) mix-ins via https://github.com/bluesky/ophyd/pull/365/files which just makes sure the python side string that ends up in stage sigs.

> First off, this should not cause a TimeoutError since AD has a PV that describes if setting the file path succeeded (`$(P):TIFF1:FilePathExists_RBV`).

This is independent of if ` yield from bps.mv(areadetector.tiff1.file_path,...)` succedes.   If setting the file path fails, but the old value exists the `FilePathExists` will still be True and if you sucessfully set the path to a non-existant path, `FilePathExists` will be False.

> Second, the user should take control of this PV (needs an interface in ophyd) and set it to `-len(new_file_path.split("/"))` (negative number and then number of subdirectories to be created).

This seems like a reasonable thing to do in `stage`.

We could make a special sub-class of `EpicsSignalWithRBV` for the file names which do the comparison ignoring the trailing slash.

It might also make sense to make a `FilePathDevice` device that wraps up all of the file-path related PVs into one thing.

prjemian, 2019-05-24:
+1 for the `FilePathDevice`

This *is* a special case worthy of its own handling since there is no timeout from the control system.  The status PV responds promptly with Ok/Fail indication. 

There can be a reason not to automatically set the value to the number of specified subdirectories.  A confused user (or a typo) might create an unwanted number of directories.  I believe that is why the `CreateDirectory` is designed thus.

klauer, 2019-05-28:
I think at least a subclass of `EpicsSignal(WithRBV?)` should exist for path handling. `FilePathDevice` sounds interesting - not sure about the added benefit vs complexity, though.

prjemian, 2019-05-28:
What other ways are there to handle the situation like this where one has an apparent EpicsSignalWithRBV and a status signal?

prjemian, 2019-05-28:
Possibly add status_pv attribute (and associated support) to EpicsSignalWithRBV?  The support would allow an exit before the timeout.

tacaswell, 2019-05-29:
> The status PV responds promptly with Ok/Fail indication.

Nominally, we have seen this hang for ~3 minutes due to local file system issues.

> since there is no timeout from the control system.

In this case, it is _technically_  a timeout in the control system.  You asked it to set the file path to `"/tmp/one/two/three/four/five"` and it actually set the filepath to `"/tmp/one/two/three/four/five/", the RBV will never be what it was set to, hence the timeout.   I agree that is likely not what the user wants, but I don't think it is a bug (as in everything is working correctly).  Rather it is a missing feature / extra sharp edge that the user needs to know to pass the file path with a trailing slash.  This is also admittedly very hair-split-ty. 

> not sure about the added benefit vs complexity, though

I think this is part of a long running discussion @klauer and I are having.  I see a sub-class of `EpicsSignal` that know about 3 PVs as being more complex than a `Device` that has multiple signals and a custom `set` method.  To some degree this is an implementation detail (as in `bp.mov(ad.hdf.file_path, '/some/path')` should "just work") which is why I want to blur the distinction between between `Signal` and `Device` (I also see Ken's point that knowing what the semantic (not technical) leaves of the tree are as being important).

I either way we, I think we want the following behavior:

 - set the file path set point
 - wait for the RBV to match independent of trailing slashes (which also needs to be widows path aware!).  We might be able to use `pathlib` for this.  If they don't match within a timeout, fail the status object
 - check the "path exists" light. if the path does not exist, fail the status object 

This would let us drop some of the special logic in https://github.com/bluesky/ophyd/blob/dfa0a4a78b8189497e783120950763a80948853b/ophyd/areadetector/filestore_mixins.py#L444-L446

----

@prjemian Is this "triple PV" common in AD/epics in general?

prjemian, 2019-05-29:
*Technically*, it is a failure or the ophyd EpicsSignalWithRBV device to support the information provided by EPICS area detector.  There is no timeout, EPICS has provided a response but that response does not fit the expectation of the EpicsSignalWithRBV device.

prjemian, 2019-05-29:
I believe this "triple PV" is a singleton at this time.

PR #939 (2020-12-15): add create_directory signal to standard AD support
Adds `create_directory` signal to all AD file writers to fix #728 and close #938.

Needs to be set to `None` in subclasses for plugins that do not have this feature.  Perhaps that is AD <2.2.
closed 2020-12-15 by danielballan