bluesky/ophyd #741: RuntimeWarning: coroutine 'Event.wait' was never awaited

prjemian, 2019-06-21:
We've done something but cannot figure out from this console report what to do next.

```
Out[5]: ()

/share1/bluesky/conda_envs/bluesky/lib/python3.7/site-packages/ophyd/ophydobj.py:269: RuntimeWarning: coroutine 'Event.wait' was never awaited
  cb(*args, **kwargs)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
In [6]:
```

Regarding this [APS issue](https://github.com/APS-USAXS/ipython-usaxs/issues/266), how do we resolve the root cause that led to this message?

We [know](https://github.com/APS-USAXS/ipython-usaxs/issues/266#issuecomment-504552882) what we were doing (measuring area detector image when the APS storage ring dumped which probably tripped our suspender for low beam) and what versions of the software we were using:

```
In [51]: h.start["versions"]                                                                                                                                                                     
Out[51]: 
{'bluesky': '1.5.3',
 'ophyd': '1.3.3',
 'databroker': '0.13.0',
 'apstools': '1.1.6',
 'epics': '3.3.3'}
```

We've seen this error at other times when the beam has not dumped (such as sometimes happens when exiting the ipython session), so we suspect it is caused by [our USAXS instrument setup](https://github.com/APS-USAXS/ipython-usaxs/tree/master/profile_bluesky/startup) and not caused by the beam dump.  

We need more advice where to look.

danielballan, 2019-06-21:
Note to self:

https://github.com/bluesky/bluesky/blob/master/bluesky/suspenders.py#L136

prjemian, 2019-06-24:
[Possibly](https://github.com/APS-USAXS/ipython-usaxs/issues/266#issuecomment-505044264) resolved in bluesky/bluesky#1125.

danielballan, 2020-03-16:
Closing now that bluesky/bluesky#1125 is released in bluesky 1.6.0.

closed 2020-03-16 by danielballan