bluesky/ophyd #757: Revisit 'another set in progress'

danielballan, 2019-08-14:
Some ideas in https://github.com/bluesky/ophyd/pull/755 were deferred in favor of simpler approach that could be back-ported to v1.3.x. Let's visit them.

prjemian, 2020-11-03:
As alluded to in [another PR](https://github.com/bluesky/ophyd/pull/901#issuecomment-716856378), the `set()` in progress is probably hanging because it has no timeout.  The way to cancel that without quitting is to call:

    st._finished()

Add kwarg `success=False` as you choose.

Isn't this the simplest solution if a timeout is not set?

The simplest solution is to always specify a finite timeout.  Be prepared to handle an exception if the timeout is reached.

physwkim, 2021-01-14:
"Another `set()` call is still in progress problem" in `EpicsSignal.set` function  seems to occur especially when it is a transition type signal. 

If 1 is set to `xxx:m1.STUP`, it changes in the following order: `0(OFF) -> 1 (On) -> 2 (BUSY) -> 0 (OFF)`.

```
xxx:m1.STUP                    2021-01-14 17:10:52.570557 OFF
xxx:m1.STUP                    2021-01-14 17:10:52.570557 ON
xxx:m1.STUP                    2021-01-14 17:11:01.923591 BUSY
xxx:m1.STUP                    2021-01-14 17:11:01.923837 OFF
```

Another transition type signal may change in the order of `0 -> 1 -> 0`.
```
test:signal                    2021-01-14 16:43:21.773892 0
test:signal                    2021-01-14 16:43:23.363370 1
test:signal                    2021-01-14 16:43:23.364933 0
```

In this case, the transition time from 1 to 0 is only 1.56 ms. The while loop in the `ophyd.utils.epics_pvs.set_and_wait` function seems difficult to detect this change.

On `camonitor`, the set was successful, but the `EpicsSignal.set` function only sees 0 values after the transition ends. Therefore, I think this type of EpicsSignal should be handled specially.

Could you please review how this solution is?
[https://github.com/physwkim/ophyd/commit/2a3b1c09c62ebf9fa5eb4807ec9e32221276edf6]()