bluesky/ophyd #759: other areadetector file plugins need warmup() method

prjemian, 2019-09-09:
A `warmup()` method was implemented for the HDF5 plugin: https://github.com/bluesky/ophyd/blob/19269911c64594ca85e27c01dd0bfd86122b8400/ophyd/areadetector/plugins.py#L889-L903

As noted in https://github.com/bluesky/ophyd/issues/598#issuecomment-420327292 (merged in #599), other file-saving plugins need this capability, too.  We've encountered this a couple times at APS (thanks to @KedoKudo for reporting this) with a TIFF plugin and also with a custom file-saving plugin.  Should be the existing `warmup()` method move to the `FilePlugin` class?

tacaswell, 2019-09-10:
There has to be a better way to do this.

danielballan, 2019-09-17:
Does anyone have time to find one? If not, I think we should move forward with this as a stop gap.

KedoKudo, 2019-09-17:
I think the warmup() method should check the port flow map (if there is one) to make sure that it can actually get an image from the camera plugin.  There could be situations where the file plugin is pointing to another processing plugin, which happen to be disconnected from the camera.