bluesky/ophyd #765: When is next release?

prjemian, 2019-10-11:
Using ophyd 1.3.3 (note its [milestone](https://github.com/bluesky/ophyd/milestone/15) is still open) and keep seeing this nuisance message:

```
/path/to/anaconda/envs/bluesky/lib/python3.7/site-packages/epics/pv.py:110: UserWarning: The `context` kwarg for epics.get_pv() is deprecated. New PVs will _not_ be created in the requested context.
  'The `context` kwarg for epics.get_pv() is deprecated. New PVs '
```

Isn't this fixed in a [later tag](https://github.com/bluesky/ophyd/releases/tag/v1.4.0rc2) but no official release or conda packages for that yet.  After removing conda package and installing with pip, subsequent `conda update` wants to revert back to this version.

The [1.4.x milestone](https://github.com/bluesky/ophyd/milestone/11) has no open issues.  Can there be an official release of 1.4.0 soon, before the end of October?

danielballan, 2019-10-11:
The only critical blocking issue I know of is to actually try to _use_ ophyd 1.4.0rc2 at a beamline and check that it works in practice. If you have an opportunity to `pip install --pre ophyd` somewhere and check that the IPython profile loads and you can run some scans, I think we can declare victory. Given the large internal changes, I'm sure there will be follow-up bug fix releases to come. (I know there are some ongoing concerns around getting the callbacks exactly right -- cc @hhslepicka.) But for a first 1.4.0 release we'd just like to do _some_ validation on the floor to avoid releasing something unusable.

A less critical issue is that we added some methods since the last ophyd release that we have recently re-considered (see https://github.com/bluesky/ophyd/issues/752) and would like to change in a backward-incompatible way. It would have been to implement that re-thinking before the initial draft made it into a release so as to avoid making a breaking change in 1.5.0. But, given our limited development bandwidth at the moment, I think the pragmatic thing to do is to add warnings to the new methods (something like "will change in a future release"...) and let it go out as is for 1.4.0.

On a separate but related subject, the situation is the same for bluesky: if you have a chance to `pip install --pre bluesky` and take 1.6.0rc2 for the spin, that can be released too. We've just been too oversubscribed with conference travel and beamline support tasks to find the time to do this yet.

danielballan, 2019-10-20:
@prjemian Have you had a chance to take the release candidates for a spin? If there isn't a good opportunity at APS either we can try to arrange something with a beamline or two at NSLS-II this week.

prjemian, 2019-10-20:
Our *opportunities* keep getting subverted by operations.  Can you test?

danielballan, 2019-10-21:
OK, will try to set something up this week.

danielballan, 2019-10-22:
We have issued new release candidates (ophyd 1.4.0rc3, bluesky 1.6.0rc3) and scheduled appointments to test them at beamlines this week. Fingers crossed for final releases by Friday, or next week at the latest.

danielballan, 2019-10-25:
We tried this at a couple beamlines this week, but packaging (conda) issues took up all the time we had at the beamline. We will try again next week.

prjemian, 2020-01-07:
We used ophyd 1.4.0rc3 (and then subsequent modifications to that) at the APS XPCS instrument in November and December.  We encountered problems such as a [`FailedStatus` exception](https://github.com/aps-8id-trr/ipython-8idiuser/issues/114) such that the instrument team reverted to using SPEC.  We are working (such as https://github.com/bluesky/ophyd/pull/788) on resolving the problems encountered and testing at the XPCS instrument again during the January 2020 shutdown.

danielballan, 2020-02-24:
@prjemian is aware, but for anyone else following along:

Tests at NSLS beamlines have been going well, and usage of the RCs at LCLS and APS has been smooth. We are on track to tag in time for NSLS' next deployment, which would mean by March 16.

danielballan, 2020-03-11:
A 1.4.0 tag will be made by March 16 or earlier. Closing.

closed 2020-03-11 by danielballan