bluesky/ophyd #766: Add warnings to walk_* methods

danielballan, 2019-10-11:
See #765

klauer, 2020-03-03:
I think a replacement should be put in prior to adding warnings?

ZLLentz, 2020-03-03:
Is there a plan to remove these methods? That's seems like a mistake to me, these are extremely useful.

danielballan, 2020-03-04:
Sorry for the overly-terse issue description at the top of this issue; let me try to do better here.

We're all agreed that they are extremely useful. The plan is to rework them in a way that probably won't be backward-compatible, based on a discussion in person with @klauer and @tacaswell months ago. I attempted to record that discussion in https://github.com/bluesky/ophyd/issues/752. I No one is raising the possibility of _removing_ them, but getting the details of the signature right to generalize well to other control layers may need refinement.

I agree that warnings would be annoying. Knowing that we might change these in the next minor release, would it be sufficient to mark these as "experimental, subject to future backward-compatible changed" in the docstrings?

It would be better to get them into a stable form before the release, but in #765 we decided that shouldn't be a blocker for getting 1.4.0 final out, and I still think that is true.

PR #811 (2020-03-04): Note that walk_ instance methods are experimental.
Touches docstrings only.

Closes #766.
Related to #752 where we discuss possible changes to these methods.
closed 2020-03-04 by danielballan