bluesky/ophyd #767: add deviocstats Device

tacaswell, 2019-10-17:
@stuwilkins pointed out that many of our IOCs have https://www.slac.stanford.edu/comp/unix/package/epics/site/devIocStats/ installed.  It would be good to have an ophyd class for it.

prjemian, 2019-10-17:
Note the two long strings for Application Directory and Startup Script that are each composed of two stringout records, to be concatenated.

Application Directory = $(PREFIX):APP_DIR1 + $(PREFIX):APP_DIR2
Startup Script = $(PREFIX):ST_SCRIPT1 + $(PREFIX):ST_SCRIPT2

prjemian, 2019-10-17:
That's the implementation seen in https://github.com/epics-modules/iocStats

prjemian, 2019-10-17:
Also note that iocStats expects `PREFIX` does not have a trailing colon while the other synApps modules expect `PREFIX` to provide a trailing colon.  See https://github.com/epics-modules/xxx/issues/30#issuecomment-538673938 for more details.

prjemian, 2019-11-20:
Follow-up on my comment from last month, a workaround for the double-colon situation is to create a database of PV aliases from double-colon PV names to single-colon aliases.  This is done in the [xxx](https://github.com/epics-modules/xxx) module:

* https://github.com/epics-modules/xxx/blob/master/xxxApp/Db/iocAdminSoft_aliases.db
* https://github.com/epics-modules/xxx/blob/master/xxxApp/Db/iocAdminVxWorks_aliases.db