bluesky/ophyd #776: TimeoutError from EpicsSignal

prjemian, 2019-11-25:
One of the chronic problems that plague the APS beam lines from relying on Bluesky for user operations is `TimeoutError: Failed to read {some PV} within None sec` that happens, seemingly at random.  When this happens, the exception stops the running plans completely.  Keep in mind, such timeouts are not on initial connection but happen after innumerable previous successful EPICS CA monitor updates on the same.  For example, sometimes it fails on the engineering units of an EPICS motor (motor.EGU) that, on point number 47 of a step scan, raises a TimeoutError.  The previous steps did not raise this error.

```
In [15]: si3.hgap                                                                                                     
Out[15]: EpicsMotor(prefix='8idi:Slit3Hsize', name='si3_hgap', parent='si3', settle_time=0.0, timeout=None, read_attrs=['user_readback', 'user_setpoint'], configuration_attrs=['user_offset', 'user_offset_dir', 'velocity', 'acceleration', 'motor_egu'])

In [16]: si3.hgapCA.Client.Exception...............................................                                   
    Warning: "Virtual circuit unresponsive"
    Context: "lambda.xray.aps.anl.gov:5064"
    Source File: ../tcpiiu.cpp line 920
    Current Time: Tue Nov 19 2019 15:02:30.072242898
..................................................................
Unexpected problem with CA circuit to server "lambda.xray.aps.anl.gov:5064" was "Connection reset by peer" - disconnecting
CA.Client.Exception...............................................
    Warning: "Virtual circuit disconnect"
    Context: "lambda.xray.aps.anl.gov:5064"
    Source File: ../cac.cpp line 1225
    Current Time: Tue Nov 19 2019 15:03:52.569633839
..................................................................
In [16]: si3.hgap                                                                                                     
Out[16]: EpicsMotor(prefix='8idi:Slit3Hsize', name='si3_hgap', parent='si3', settle_time=0.0, timeout=None, read_attrs=['user_readback', 'user_setpoint'], configuration_attrs=['user_offset', 'user_offset_dir', 'velocity', 'acceleration', 'motor_egu'])

/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/epics/ca.py:1459: UserWarning: ca.get('8idi:m54.TDIR') timed out after 1.00 seconds.
  warnings.warn(msg % (name(chid), timeout))
Subscription value callback exception (EpicsSignalRO(read_pv='8idi:m54.DMOV', name='samplestage_x_motor_done_move', parent='samplestage_x', value=1, timestamp=1574696427.776781, auto_monitor=True, string=False))
Traceback (most recent call last):
  File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/ophyd/ophydobj.py", line 471, in inner
    cb(*args, **kwargs)
  File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/ophyd/device.py", line 1603, in wrapped
    ret = func(self, *args, **kwargs)
  File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/ophyd/epics_motor.py", line 246, in _move_changed
    if self.direction_of_travel.get() == 0:
  File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/ophyd/signal.py", line 830, in get
    raise TimeoutError(f'Failed to read {self._read_pvname} within '
TimeoutError: Failed to read 8idi:m54.TDIR within None sec
In [17]: si3.hgap                                                                                                     
Out[17]: EpicsMotor(prefix='8idi:Slit3Hsize', name='si3_hgap', parent='si3', settle_time=0.0, timeout=None, read_attrs=['user_readback', 'user_setpoint'], configuration_attrs=['user_offset', 'user_offset_dir', 'velocity', 'acceleration', 'motor_egu'])
```

Here is an example from interactive use where the .`TDIR` (motor target direction changed sign, went the other way) raised a TimeoutError.  ironically, this field is of no use to the bluesky user so this failure is unnecessary, at best (that's an ophyd device configuration issue).  Of concern is the timeout `with None sec` and why a timeout at all.

By the way, 

```
In [21]: print(APS_utils.print_RE_md(printing=False))                                                                 
=========== ================================
key         value                           
=========== ================================
beamline_id APS TRR 8-ID-I                  
login_id    8idiuser@quartz.xray.aps.anl.gov
pid         7116                            
proposal_id None                            
scan_id     210                             
versions    =========== ========            
            key         value               
            =========== ========            
            apstools    1.1.14              
            bluesky     1.5.4               
            databroker  0.13.0              
            epics       3.4.0               
            matplotlib  3.1.0               
            numpy       1.16.4              
            ophyd       1.4.0rc1            
            pyRestTable 2020.0.2            
            spec2nexus  2021.1.3            
            =========== ========            
=========== ================================
In [22]: print(APS_utils.device_read2table(si3.hgap))                                                                 
====================== ================== ==========================
name                   value              timestamp                 
====================== ================== ==========================
si3_hgap               15.000060000000001 2019-11-19 14:00:00.379946
si3_hgap_user_setpoint 15.000000000000002 2019-11-19 14:00:00.379946
====================== ================== ==========================
```

prjemian, 2019-11-25:
*And*, another example during a scan:

```
In [1]:                                                                                                                                                                                           

In [1]: samplestage.x                                                                                                                                                                             
Out[1]: EpicsMotor(prefix='8idi:m54', name='samplestage_x', parent='samplestage', settle_time=0.0, timeout=None, read_attrs=['user_readback', 'user_setpoint'], configuration_attrs=['user_offset', 'user_offset_dir', 'velocity', 'acceleration', 'motor_egu'])

In [2]: mov scaler1.preset_time 0.1 
   ...:                                                                                                                                                                                           

In [3]: RE(bp.rel_scan([scaler1,lakeshore],samplestage.x,-2.0, 2.0, 41))                                                                                                                          
Transient Scan ID: 211     Time: 2019-11-25 10:14:47
Persistent Unique Scan ID: 'c278bc5b-76ac-4ae5-952c-7a949d036975'
New stream: 'baseline'
Start-of-run baseline readings:
+--------------------------------+--------------------------------+
+--------------------------------+--------------------------------+
New stream: 'aps_current_monitor'
New stream: 'primary'                                                                                                                                                                             
+-----------+------------+---------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+
|   seq_num |       time | samplestage_x |   Timebase |      pind1 |      pind2 | pind3_APD_ |      pind4 |       pdbs |      I_APS |      I0Mon |        APD |    cstar_l |    cstar_h |     oxygen |
+-----------+------------+---------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+------------+
|         1 | 10:14:54.4 |      7.584250 |    1000000 |        104 |         77 |          1 |        266 |         59 |      50979 |          2 |          1 |          2 |          0 |          1 |
|         2 | 10:14:55.6 |      7.684250 |    1000000 |        104 |         78 |          1 |        265 |         60 |      50964 |          2 |          1 |          2 |          0 |          1 |
|         3 | 10:14:56.7 |      7.784250 |    1000000 |        104 |         78 |          1 |        266 |         59 |      50960 |          1 |          1 |          1 |          0 |          1 |
|         4 | 10:14:57.9 |      7.884250 |    1000000 |        104 |         78 |          1 |        266 |         60 |      50954 |          3 |          1 |          3 |          0 |          1 |
|         5 | 10:14:59.0 |      7.984250 |    1000000 |        104 |         78 |          1 |        265 |         60 |      50954 |          2 |          1 |          2 |          0 |          1 |
|         6 | 10:15:00.2 |      8.084250 |    1000000 |        104 |         78 |          1 |        265 |         59 |      50953 |          4 |          1 |          5 |          1 |          1 |
|         7 | 10:15:01.3 |      8.184250 |    1000000 |        104 |         78 |          1 |        265 |         60 |      50952 |          0 |          1 |          0 |          0 |          1 |
|         8 | 10:15:02.5 |      8.284250 |    1000000 |        104 |         78 |          1 |        265 |         60 |      50952 |          3 |          1 |          3 |          0 |          1 |
|         9 | 10:15:03.6 |      8.384250 |    1000000 |        104 |         78 |          1 |        265 |         59 |      50952 |          0 |          1 |          0 |          0 |          1 |
|        10 | 10:15:04.7 |      8.484250 |    1000000 |        104 |         78 |          1 |        265 |         59 |      50952 |          0 |          1 |          0 |          0 |          1 |
/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/epics/ca.py:1459: UserWarning: ca.get('8idi:m54.EGU') timed out after 1.00 seconds.
  warnings.warn(msg % (name(chid), timeout))
[E 10:15:08.818 run_engine:1261] Run aborted                                                                                                                                                      
    Traceback (most recent call last):
      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/bluesky/run_engine.py", line 1150, in _run
        msg = self._plan_stack[-1].send(resp)
      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1294, in __call__
        return (yield from plan)
      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1149, in baseline_wrapper
        return (yield from plan_mutator(plan, insert_baseline))
      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 169, in plan_mutator
        raise ex
      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/bluesky/preprocessors.py", line 122, in plan_mutator
[... cut ...]
    828         if info is None:
    829             timeout = kwargs.get('timeout', None)
--> 830             raise TimeoutError(f'Failed to read {self._read_pvname} within '
    831                                f'{timeout} sec')
    832 

TimeoutError: Failed to read 8idi:m54.EGU within None sec

In [4]:                                                                                                                                                                                           
```

danielballan, 2019-11-25:
I suspect this is a bug in ophyd, pyepics, or more specifically the integration between the two (which lives in ophyd). I notice you are running on an ophyd prerelease, `1.4.0rc1`. Have you ever seen this on a version in the 1.3.x series?

prjemian, 2019-11-25:
Yes.  It's time to document this and catch the root cause

prjemian, 2019-11-25:
Timeout=None sounds like it comes from ophyd, agree?

prjemian, 2019-11-26:
BUT, this line in the error dump:

      File "/home/beams/8IDIUSER/.conda/envs/bluesky/lib/python3.7/site-packages/ophyd/epics_motor.py", line 246, in _move_changed

calls `.get()` (from `if self.direction_of_travel.get() == 0:`) from the underlying pyepics support.  That support should supply a valid default for the `timeout` kwarg according to this documentation: *timeout: maximum time to wait for value to be received.  (default = `0.5 + log10(count) seconds`)*

Within PyEpics, there are PV objects and Motor objects.  This is a PV object.  The call chain appears to be:
  - [PV.get()](https://github.com/pyepics/pyepics/blob/master/epics/pv.py)     # (no `timeout` kwarg, becomes  `None` by default)
  - -> PV.get_with_metadata()     # (`timeout None` input)
  - -> [ca.get_with_metadata()](https://github.com/pyepics/pyepics/blob/master/epics/ca.py)     # (`timeout=None` input)
  - -> ca.get_complete_with_metadata()    #  (`timeout=None` input)

Here, `timeout is None` is handled:

    if timeout is None:
        timeout = 1.0 + log10(max(1, count))

and then watches for a timeout.

Is there code in ophyd that raises the TimeoutError exception?  Not seeing that exception from PyEpics.  see this table:

repo | `TimeoutError` in code
---- | ----
[PyEpics](https://github.com/pyepics/pyepics/search?q=TimeoutError&unscoped_q=TimeoutError) | 0 instances
[ophyd](https://github.com/bluesky/ophyd/search?q=TimeoutError&unscoped_q=TimeoutError) | 17 instances

Time to look for the relevant TimeoutError in ophyd.

prjemian, 2019-11-26:
`TimeoutError` not raised in ophyd/epics_motor.py, only passed along.  It comes from this set of lines: https://github.com/bluesky/ophyd/blob/9da7884fb764327d2ba3532302b75bb49b7e5872/ophyd/signal.py#L824-L831

In the wrong set of circumstances that trigger *our* problem, the `info` object is set to `None`.  Then, on this line: `timeout = kwargs.get('timeout', None)` the timeout (not specified) will be reported incorrectly as `None`.  Shouldn't it be reported based on the actual time waited (since PyEpics has default for `timeout=None`)?  Could be done with an actual timer here.

The *value* of the timeout is not the real problem, though, only a symptom.  Why do we get a timeout at all?  First off, though, we don't know how long we have waited.

prjemian, 2019-11-26:
Actually, why is `None` being returned from `self._read_pv.get_with_metadata()`?  PyEpics has called [warnings](https://docs.python.org/3/library/warnings.html) with this code:

        if time.time()-t0 > timeout:
            msg = "ca.get('%s') timed out after %.2f seconds."
            warnings.warn(msg % (name(chid), timeout))
            return None

This code should handle a possible result from `warnings`.  One way is outlined in the [SO post](https://stackoverflow.com/a/5645133/1046449)  I suspect this change will help to reveal the underlying problem where `self._read_pv.get_with_metadata()` returns `None`.

prjemian, 2019-11-26:
Here's the simple example from SO:

```
import warnings

def fxn():
    warnings.warn("deprecated", DeprecationWarning)

with warnings.catch_warnings(record=True) as w:
    # Cause all warnings to always be triggered.
    warnings.simplefilter("always")
    # Trigger a warning.
    fxn()
    # Verify some things
    assert len(w) == 1
    assert issubclass(w[-1].category, DeprecationWarning)
    assert "deprecated" in str(w[-1].message)
```

prjemian, 2019-11-26:
Using `warnings.catch_warnings(record=True)` comes at a [price we cannot pay](https://docs.python.org/3/library/warnings.html#available-context-managers) while waiting for EPICS PV values to come in from the network environment:

> The `catch_warnings()` manager works by replacing and then later restoring the module’s `showwarning()` function and internal list of filter specifications. This means the context manager is modifying global state and therefore is not thread-safe.

prjemian, 2019-11-26:
Perhaps converting these warnings to `UserWarning` exceptions is thread-safe:

code:

```
import warnings

def function1():
    for action in ("error", "default"):
        print("-"*20)
        warnings.filterwarnings(action)
        try:
            function2()
            print(f"no exception from action {action}")
        except UserWarning as exc:
            print(f"caught: {exc}")

def function2():
    warnings.warn("example warning message")

function2()
function1()
```

output:

```
In [2]: function2()                                                                                                                                                     
/home/mintadmin/Apps/anaconda/envs/db-test/bin/ipython:14: UserWarning: example warning message

In [3]: function1()                                                                                                                                                     
--------------------
caught: example warning message
--------------------
/home/mintadmin/Apps/anaconda/envs/db-test/bin/ipython:14: UserWarning: example warning message
no exception from action default

In [4]:                                                                                                                                                                 
```

prjemian, 2019-11-26:
One possible revision of https://github.com/bluesky/ophyd/blob/9da7884fb764327d2ba3532302b75bb49b7e5872/ophyd/signal.py#L824-L831 is as follows:

```python
    self.wait_for_connection(timeout=connection_timeout) 
    try:
        warnings.filterwarnings("error")
        t0 = time.time()
        info = self._read_pv.get_with_metadata(
                    as_string=as_string, form=form, **kwargs) 
    except UserWarning as exc:
        print(f"while getting from {self._read_pvname}, caught '{exc}'")
    finally:
        dt = time.time() - t0
        warnings.filterwarnings("default")
    
    if info is None: 
        timeout = kwargs.get('timeout', None) 
        raise TimeoutError(f'Failed to read {self._read_pvname} in ' 
                           f'{dt:.3f} sec (with timeout={timeout})') 
```

prjemian, 2019-12-10:
As commented for [APS 8-ID-I XPCS](https://github.com/aps-8id-trr/ipython-8idiuser/issues/114#issuecomment-564116405), a ...

> combination of [connect() or get()] retries and direct ophyd handling of the various timeouts (connection, get, put) is the first hypothesis to resolve this issue.

prjemian, 2019-12-10:
@kmpeters [reported](https://github.com/aps-8id-trr/ipython-8idiuser/issues/46#issuecomment-562722515) similar experience with PyEpics (not bluesky):

> I've seen `TimeoutError` exceptions in fly-scan python code using pyepics running on Windows at (APS) 34ID-E. The problem occurred infrequently (less frequently than once every few weeks). The PV that was being accessed at the time of the timeout would change each time. Sometimes the timeout was on a get and other times it was on put. The PV that was being accessed when the exception occurred was always available via the command line immediately after the exception occurred. I wrote safer put and get functions that handled the timeout exception and printed info about the problem and I haven't heard from users about the problem since then.
> 
> I don't remember the python, pyepics3, and EPICS lib versions. I can look them up on the next shutdown day.

prjemian, 2019-12-10:
@kmpeters - Can you post a snippet of the safer `put()` and `get()` codes?  That would make a great starting point for this issue.

kmpeters, 2019-12-10:
It looks like the latest version of my fly-scan code only has a safer ``put()``:
```python
	#
	# Debugging
	#

	def debugPut(self, pv, val, wait=False, use_complete=False, timeout=30.0):
		#
		exitFlag = False

		while ( not exitFlag ):
			try:
				pv.put(val, wait=wait, use_complete=use_complete, timeout=timeout)
			except WindowsError as e:
				print
				print "EPICS put FAILED:"
				print "\tPV =", pv.pvname
				print "\tval =", val
				print "\twait =", wait, "; use_complete =", use_complete, "; timeout =", timeout
				print "\tconnected =", pv.connected, "; host =", pv.host
				print
				print "A WindowsError occurred:"
				# The following line results in an AttributeError. Apparently e.type doesn't exist
				#!print "\t", e.type
				print "\t", e
				print "\t", e.args
				print
				print "Backtrace:"
				traceback.print_stack()
				print
				time.sleep(15.00)
			else:
				exitFlag = True
```

kmpeters, 2019-12-10:
The safer ``get()`` function I was remembering was much older.  It was used to handle exceptions on read timeouts using CaChannel.

prjemian, 2019-12-12:
@kmpeters : Looking over the code you supplied to make a safer `put()`, it seems somewhat similar to the method I described [above](https://github.com/bluesky/ophyd/issues/776#issuecomment-558857972), which converts the `warnings` posted by PyEpics into an actual `UserError` which can be trapped.  In your code, a `WindowsError` is trapped.

I'll begin testing the idea of trapping a `warnings` call as an exception (`UserError`).

prjemian, 2019-12-19:
@kmpeters: Trapping the `warnings` call from PyEpics as a `UserException` was [successful](https://github.com/aps-8id-trr/ipython-8idiuser/issues/114#issuecomment-567648403).  Then, implementing the `safer_get()` including retries with shorter, but escalating, timeout was [successfule](https://github.com/aps-8id-trr/ipython-8idiuser/issues/114#issuecomment-567700053) as well.  Thanks, again, for the assist!

tacaswell, 2019-12-20:
I think `bandaid_get` is a better name than `safer_get`.  I left a long comment over at https://github.com/aps-8id-trr/ipython-8idiuser/issues/114#issuecomment-567945638 about why I am not sure re-trying is the best solution.

prjemian, 2019-12-20:
The first item to resolve in this issue is to capture the `warnings` content from PyEpics when it is posted.  I'll file a PR on this specifically as resolution to this issue.  That won't stop the `FailedStatus` exception noted in https://github.com/aps-8id-trr/ipython-8idiuser/issues/114 but at least we'll know for sure that PyEpics has given us a timeout.

prjemian, 2019-12-20:
note:  not seeing this on my VM, need to try at APS beam line 8-ID-I, it seems

test code: https://github.com/prjemian/bstesting/blob/master/test4.py

test database when not at APS 8-ID-I: https://github.com/prjemian/bstesting/blob/master/8idi.db

    softIoc -d 8idi.db

prjemian, 2020-01-27:
With #788, I'm declaring victory here and closing the issue as **SOLVED**.

closed 2020-01-27 by prjemian