bluesky/ophyd #782: motor motion fails to complete #1

prjemian, 2019-12-13:
As reported https://github.com/prjemian/ipython-vm7/issues/1, 

> Sometimes, motor motion fails to complete. First noticed that recently when testing `bec.LivePlot` problems for the APS USAXS instrument, running a specific sequence of activities within an infinite loop. At random time, the sequence waits indefinitely for the motor. The motor is an EPICS soft motor record running in a self-hosted linux VM on the same workstation as the bluesky session.
> 
> In EPICS, the motor has reached position and the DMOV is 1. In bluesky, the move is waiting to complete but has reached 100%.

This test code replicates the problem (and assume there is an EPICS motor record `sky:m1` available):

```
#!/usr/bin/env python

from bluesky import RunEngine
from bluesky import SupplementalData
from bluesky.callbacks.best_effort import BestEffortCallback
from bluesky.utils import get_history
from bluesky.utils import ProgressBarManager
import bluesky.plan_stubs as bps
from databroker import Broker
import epics
import logging
from ophyd import EpicsMotor
import time

# logging.basicConfig(level=logging.DEBUG)
REPETITIONS = 10
CYCLES = 250
DELAY_S = 1e-6

bec = BestEffortCallback()
db = Broker.named("mongodb_config")
sd = SupplementalData()
pbar_manager = ProgressBarManager()

RE = RunEngine(get_history())
# RE.subscribe(db.insert)
RE.subscribe(bec)
# RE.preprocessors.append(sd)
# RE.waiting_hook = pbar_manager

m1 = EpicsMotor("sky:m1", name="m1")
while not m1.connected:
    time.sleep(.05)

def move(motor, label, dest, delay_s):
    yield from bps.checkpoint()
    yield from bps.mv(m1, dest)
    print(f"{label}:   {dest} {motor.position}")
    yield from bps.sleep(delay_s)

def ping_pong(motor, v1, v2, cycles=100, delay_s=1e-2):
    for i in range(cycles):
        yield from move(m1, f"ping {i+1}", v1, delay_s)
        yield from move(m1, f"pong {i+1}", v2, delay_s)

if __name__ == "__main__":
    RE(ping_pong(m1, 1, -1, cycles=CYCLES, delay_s=DELAY_S))
```

prjemian, 2019-12-13:
We've been having a robust discussion on the [Slack/Nikea channel](https://nikea.slack.com/archives/C7PNB3BUG/p1576219332054700).  @tacaswell summarized what *should* happen:

> the chain here is that:
> 
> *    The set method returns a StatusBase object and keeps a reference to it
> *    The RE creates a the asyncio.Event registers a function to set it with the StatusBase object and keeps a reference to it
> *    On wait the RE waits for the asyncio.Event to be set
> *    When the st._finished() method is called, the callbacks are run and the asyncio.Event is released

prjemian, 2019-12-13:
Russ Berg, Canadian Light Source, [reported](https://nikea.slack.com/archives/C7PNB3BUG/p1576249451022900) finding this same issue yesterday:

>  I literally just ran into this yesterday when creating a motor class for one we run here at the CLS and the issue (near as I can figure) was that the subscription for the callback to return that the motor had stopped and the request to "put" the new setpoint created a race condition such that the callback would fire before the motor had started to move, which caused the scan to either report multiple stops (causing multiple scan requests to move to next setpoint) or the scan would simply be left waiting for the motor to stop (which it already had). The solution that worked for me was to put a small sleep in between the "put" and the subscription to the callback. here is a paste oy my move function that worked finally 

```
    @raise_if_disconnected
    def move(self, position, wait=True, **kwargs):
        """
        Set the setpoint and return a Status object that monitors the 'done' PV.
        """
        #print('in reixs.devices.clsEpicsMotor implementation of move()')
        status = DeviceStatus(self)        # Wire up a callback that will mark the status object as finished
        # when the done signal goes from low to high---that is, a positive edge.
        def callback(old_value, value, **kwargs):
            if (value == 0):
                self.motor_is_moving.clear_sub(callback)
                status._finished()
                status.done = True        # Now 'put' the value.        # THE ORDER OF THE FOLLOWING CALLS IS SUPER CRITICAL, THE MOVE NEEDS TO START BEFORE YOU SUBSCRIBE OTHERWISE
        # THERE WILL BE MULTIPLE MOVES REQUESTS PER STOP
        self.user_setpoint.put(position, wait=True)
        time.sleep(0.15)
        self.motor_is_moving.subscribe(callback)        # And return the Status object, which the caller can use to
        # tell when the action is complete.
        return status
```

@tacaswell responded:

> There should be de-bouncing logic for handling this case in the EpicsMotor / Positioner classes
> we should be able to solve this without putting sleep calls in (as they will mostly over-wait until the day when they are not long enough)

prjemian, 2019-12-13:
Looking through Russ's code, making the status object in the move command seems right.  Subscribe the callback first to the self.motor_done_move (the motor's .DMOV field), then move the motor.  That's the order I think is normal for EPICS clients.  We need to avoid the multiple move requests associated with the .MOVN field.  The logic for the callback field will need adjusting since .DMOV=1 when the motor is not moving.

This needs to be tested for the case when the motor is re-targeted while it is moving.  That should not create a new status object but it should call motor.user_setpoint.put(new_destination).

prjemian, 2019-12-13:
Russ [responds](https://nikea.slack.com/archives/C7PNB3BUG/p1576270226044900):

> here at the CLS we have a motor driver that was developed in house years ago and is being used on a lot of our beamlines, it isnt using the standard Epics motor record/driver so its processing seems to be a bit different when it comes to the overall status, as a result the only solution that worked for it (at this point in time anyway) is the delay between "put" and "subscribe", I tried the approach you mention above and as you can see from the output the scan appears to see the motor stop so it issues the next setpoint but the motor is just sitting there

```
+-----------+------------+------------+-------------------------+------------+
|   seq_num |       time |  DET_TRANS | DET_TRANS_user_setpoint |  noisy_det |
+-----------+------------+------------+-------------------------+------------+
|         1 | 14:43:29.3 |      8.175 |                  -8.000 |      1.094 |
|         2 | 14:43:40.4 |     -4.477 |                  -4.000 |      1.039 |
|         3 | 14:43:40.4 |     -4.477 |                   0.000 |      1.075 |
|         4 | 14:43:40.4 |     -4.477 |                   4.000 |      0.941 |
|         5 | 14:43:40.4 |     -4.477 |                   8.000 |      0.902 |
+-----------+------------+------------+-------------------------+------------+
```

prjemian, 2019-12-13:
With this replacement for the `ophyd.epics_motor.EpicsMotor.move()` method, was able to run the `ping_pong()` plan for 250 cycles with no stall.

```
    @raise_if_disconnected
    def move(self, position, wait=True, **kwargs):
        '''Move to a specified position, optionally waiting for motion to
        complete.

        Parameters
        ----------
        position
            Position to move to
        moved_cb : callable
            Call this callback when movement has finished. This callback must
            accept one keyword argument: 'obj' which will be set to this
            positioner instance.
        timeout : float, optional
            Maximum time to wait for the motion. If None, the default timeout
            for this positioner is used.

        Returns
        -------
        status : MoveStatus

        Raises
        ------
        TimeoutError
            When motion takes longer than `timeout`
        ValueError
            On invalid positions
        RuntimeError
            If motion fails other than timing out
        '''
        self._started_moving = False
        status = None

        def dmov_callback(old_value, value, **kwargs):
            self.log.debug(f'DEBUG: dmov_callback(): new={value} old={old_value}')
            if self._started_moving and value == 1:
                self.motor_done_move.clear_sub(dmov_callback)
                status._finished(success=True)
                status.done = True

        self.motor_done_move.subscribe(dmov_callback)
        status = super().move(position, **kwargs)

        self.user_setpoint.put(position, wait=False)
        try:
            if wait:
                status_wait(status)
        except KeyboardInterrupt:
            self.stop()
            raise

        return status
```

prjemian, 2019-12-13:
I'll post this change in a branch with a PR.

mrakitin, 2019-12-13:
cc @RussBerg

prjemian, 2019-12-19:
Try implementing the changes from https://github.com/aps-8id-trr/ipython-8idiuser/issues/114#issuecomment-567703249 with the new example VM.

prjemian, 2019-12-19:
These changes in `ophyd.signal.EpicsMotorBase.get()`:

<details>
<summary>changes</summary>
<p>

```python
    def prj_get(self, *, as_string=None, form='time', 
                max_wait_time_s=3.0,
                max_retries=10,
                **kwargs):
        counter = 0
        retry_timeout = kwargs.get('timeout', 0.05)
        scale_factor = max_wait_time_s / exp(max_retries-1)  # t seconds for n retries

        t0 = time.time()
        warnings.filterwarnings("error")
        while time.time() - t0 < max_wait_time_s:
            # What if previously-connected PV has disconnected?
            # Do we need to check here or does PyEpics raise a warning or exception for this?
            try:
                info = self._read_pv.get_with_metadata(
                            as_string=as_string, form=form, 
                            timeout=retry_timeout, **kwargs)
                if counter > 0:
                    dt = time.time() - t0
                    self.log.debug('get({}) pv={} retries={} in {:.3f} sec'
                                .format(self.name, self._read_pvname, counter, dt))
                return info
            except UserWarning as exc:
                # timeout = kwargs.get('timeout', None)
                # raise TimeoutError('Failed to read {} within: {} sec: {}'
                #                 .format(self._read_pvname, timeout, exc))
                info = None
                print(counter, exc, f"({time.time()-t0:.2f}s)")
            finally:
                counter += 1
                retry_timeout = min(
                    max_wait_time_s - (time.time() - t0),        # limit to max_wait_time_s
                    retry_timeout + scale_factor * exp(counter)  # progressively longer wait time
                )

        dt = time.time() - t0
        warnings.filterwarnings("default")
        if counter > 0:
            self.log.debug('get({}) pv={} retries={} in {:.3f} sec'
                           .format(self.name, self._read_pvname, counter, dt))
        if info is None:
            raise TimeoutError('Failed to read {} within: {} sec'
                               .format(self._read_pvname, dt))

    def get(self, *, as_string=None, connection_timeout=1.0, form='time',
            **kwargs):
        '''Get the readback value through an explicit call to EPICS

        Parameters
        ----------
        count : int, optional
            Explicitly limit count for array data
        as_string : bool, optional
            Get a string representation of the value, defaults to as_string
            from this signal, optional
        as_numpy : bool
            Use numpy array as the return type for array data.
        timeout : float, optional
            maximum time to wait for value to be received.
            (default = 0.5 + log10(count) seconds)
        use_monitor : bool, optional
            to use value from latest monitor callback or to make an
            explicit CA call for the value. (default: True)
        connection_timeout : float, optional
            If not already connected, allow up to `connection_timeout` seconds
            for the connection to complete.
        form : {'time', 'ctrl'}
            PV form to request
        '''
        if as_string is None:
            as_string = self._string

        self.wait_for_connection(timeout=connection_timeout)
        info = self.prj_get(**kwargs)
        # try:
        #     warnings.filterwarnings("error")
        #     t0 = time.time()
        #     info = self._read_pv.get_with_metadata(
        #                 as_string=as_string, form=form, **kwargs) 
        # except UserWarning as exc:
        #     timeout = kwargs.get('timeout', None)
        #     raise TimeoutError('Failed to read {} within: {} sec: {}'
        #                        .format(self._read_pvname, timeout, exc))
        # finally:
        #     dt = time.time() - t0
        #     warnings.filterwarnings("default")

        if info is None:
            timeout = kwargs.get('timeout', None)
            raise TimeoutError('Failed to read {} within: {} sec}'
                               .format(self._read_pvname, timeout))

        value = info.pop('value')
        if as_string:
            value = waveform_to_string(value)

        has_monitor = self._monitors[self.pvname] is not None
        if form != 'time' or not has_monitor:
            # Different form such as 'ctrl' holds additional data not available
            # through the DBR_TIME channel access monitor
            self._metadata_changed(self.pvname, info, update=True,
                                   require_timestamp=True, from_monitor=False)

        if not has_monitor:
            # No monitor - readback can only be updated here
            self._readback = value
        return self._fix_type(value)
```

</p>
</details>

tacaswell, 2019-12-20:
@prjemian can you post diffs instead of the full code?  At least I find it much easier to read and `git diff` will give you something copy-pastable.

Opening draft PRs with the changes is also a really good way to communicate what changes you had to make.

prjemian, 2019-12-20:
For sure.  So far, was just testing the basic premise.  Still, it would have been good to capture that in a branch.

prjemian, 2020-01-27:
Let's test that the fix from #776 will resolve this issue as well.  Testing at APS 8-ID-I XPCS instrument overnight.  Hoping to declare victory.

prjemian, 2020-01-27:
Running 100,000 cycles, should take about 15 hours.

prjemian, 2020-01-28:
Completed in just under 14 hours, 2% of the moves took longer than 1s to complete, the longest just under 2.5s.  A maximum allowed timeout of 2.5 would be the bare minimum, 10 s would have very low probability of occurrence.

This was using these versions:

* bluesky 1.6.0rc2+52.g1231a17  (HEAD of current master)
* ophyd 1.4.0rc4
* epics 3.4.0

Declaring victory.

commit 33e3d386 (2019-12-13): fixes #782
closed 2020-01-28 by prjemian