bluesky/ophyd #796: dotted_name() should include name of the obj

prjemian, 2020-02-08:
Something's missing in the output of the `obj.dotted_name` property -- the name of the `obj` itself.  For example, given an instance of EpicsMotor:

```
In [11]: import ophyd                                                                                                                       

In [12]: ophyd.__version__                                                                                                                  
Out[12]: '1.4.0rc4'

In [13]: m1.low_limit_switch.dotted_name                                                                                                    
Out[13]: 'low_limit_switch'
```

Expected to receive `'m1.low_limit_switch'`

tacaswell, 2020-02-08:
See https://github.com/bluesky/ophyd/pull/797#issuecomment-583771429 for a long explanation.  This is the expected behavior.

Consider:
```
a = b = c = m1
a.low_limit_switch.dotted_name # what do you expect this to be?
```

prjemian, 2020-02-08:
I expect it to be `m1.low_limit_switch`.  Don't like this explanation very much but I understand your point of view.

closed 2020-02-08 by prjemian

tacaswell, 2020-02-09:
What are you trying to do where you want to mix the two categories of names?

prjemian, 2020-02-09:
Reporting.  Nothing active. The report is a diagnostic. The full dotted
name is the most accurate description of the object in use, from the
perspective of how it is used in the ipython session. No bother here, the
report function can construct what is needed.

You asked what object name did I expect to see prefixed.  My answer was not
precise given your example

a = b = c = m1
a. attribute.dotted_name()

I expect to see the component's name field prefixed (the name when it was
defined).