bluesky/ophyd #802: motor limits updated outside of bluesky

prjemian, 2020-02-14:
Sometimes, we change our motor limits from other software while a bluesky session is running.  In such cases, ophyd does not learn that these limits have changed.  (See https://github.com/APS-USAXS/ipython-usaxs/issues/331 for example.)  We can also have procedures that will change the limits from a bluesky plan.  In those cases where we know a motor's limits will be adjusted programmatically, we subclass `EpicsMotor` and connect with the motor's `.HLM` and `.LLM` fields.  Same problem applies, ophyd does not learn of the new limits.

Is it true that ophyd's `EpicsSignalBase` only learns the limit values from the PyEpics PV metadata at the initial connection?  Is there any code that can be called to learn of updates?

prjemian, 2020-02-14:
The motor limit fields (`.HLM` and `.LLM`) will post monitors if they are changed:

```
(base) mintadmin@mint-vm:~/.../eclipse/ipython-usaxs$ camonitor sky:m1.{H,L}LM
sky:m1.HLM                     2020-02-14 15:04:32.874334 32000  
sky:m1.LLM                     2020-02-14 15:04:32.874334 -32000  
sky:m1.HLM                     2020-02-14 15:11:27.791289 100  
sky:m1.HLM                     2020-02-14 15:11:27.791314 200  
sky:m1.HLM                     2020-02-14 15:11:36.276677 200  
sky:m1.HLM                     2020-02-14 15:11:36.377094 32000  
sky:m1.HLM                     2020-02-14 15:11:40.930856 32000  
```

prjemian, 2020-02-14:
Here's our code to support changing the motor limits from ophyd:  https://github.com/BCDA-APS/apstools/blob/24b40c08a6e3fbb26a75ead306e057d92a4be417/apstools/devices.py#L1140-L1196

The `get_lim()` and `set_lim()` methods are for SPEC compatibility.

prjemian, 2020-02-14:
Seems this could be implemented easily by subscribing each of these fields to a method that updates either `self._metadata['lower_ctrl_limit']`, or `self._metadata['upper_ctrl_limit']`, as appropriate.

prjemian, 2020-02-14:
This mixin class works but there must be something better:

```
class Mixin(EpicsMotor):
    high_limit_value = Component(EpicsSignal, ".HLM", kind="omitted")
    low_limit_value = Component(EpicsSignal, ".LLM", kind="omitted")

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

        def cb_low_limit(value, old_value, **kwargs):
            if self.connected and old_value is not None and value != self.low_limit:
                print(f"new low limit:{value} old:{old_value}, kwargs={kwargs}")
                self.user_setpoint._metadata['lower_ctrl_limit'] = value
        def cb_high_limit(value, old_value, **kwargs):
            if self.connected and old_value is not None and value != self.high_limit:
                print(f"new high limit:{value} old:{old_value}, kwargs={kwargs}")
                self.user_setpoint._metadata['upper_ctrl_limit'] = value

        self.low_limit_value.subscribe(cb_low_limit)
        self.high_limit_value.subscribe(cb_high_limit)

class EpicsMotorSubclass(Mixin, EpicsMotor): ...

m1 = EpicsMotorSubclass('sky:m1', name='m1', labels=("motor",))
```

prjemian, 2020-02-14:
This behavior is inherited from PyEpics?  Adding a simple callback with a PyEpics `PV` object shows that the changes in the EPICS PV's limit value are not provided.  The value passed is from the initial connection.

```
In [19]: def watch(*args, **kwargs): 
    ...:     print(f"args={args}\nkwargs={kwargs}") 
    ...:                                                                                                                                                            

In [20]: em8.add_callback(watch, run_now=True)                                                                                                                      
args=()
kwargs={'pvname': 'sky:m1.VAL', 'value': 0.0, 'char_value': '0', 'status': 0, 'ftype': 20, 'chid': c_long(94222088162152), 'host': 'mint-vm:5064', 'count': 1, 'access': 'read/write', 'write_access': True, 'read_access': True, 'severity': 0, 'timestamp': 1581718021.08009, 'posixseconds': 1581718021.0, 'nanoseconds': 80090929, 'precision': 5, 'units': 'degrees', 'enum_strs': None, 'upper_disp_limit': 32000.0, 'lower_disp_limit': -32000.0, 'upper_alarm_limit': 0.0, 'lower_alarm_limit': 0.0, 'lower_warning_limit': 0.0, 'upper_warning_limit': 0.0, 'upper_ctrl_limit': 32000.0, 'lower_ctrl_limit': -32000.0, 'nelm': 1, 'type': 'time_double', 'typefull': 'time_double', 'cb_info': (1, <PV 'sky:m1.VAL', count=1, type=time_double, access=read/write>)}
Out[20]: 1

In [21]: mov m8.low_limit_value 31000                                                                                                                               
new low limit:31000.0 old:31000, kwargs={'timestamp': 1581718196.200964, 'status': <AlarmStatus.NO_ALARM: 0>, 'severity': <AlarmSeverity.NO_ALARM: 0>, 'precision': 5, 'lower_ctrl_limit': -1e+300, 'upper_ctrl_limit': 1e+300, 'units': 'degrees', 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.LLM', name='m8_low_limit_value', parent='m8', value=31000.0, timestamp=1581718196.200964, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.LLM', limits=False, put_complete=False)}

In [22]: mov m8 -0.1                                                                                                                                                
new:-0.1 old:0.0, kwargs={'timestamp': 1581718543.941546, 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=-0.1, timestamp=1581718543.941546, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}
new:0.0 old:-0.1, kwargs={'timestamp': 1581718543.941563, 'status': <AlarmStatus.NO_ALARM: 0>, 'severity': <AlarmSeverity.NO_ALARM: 0>, 'precision': 5, 'lower_ctrl_limit': -32000.0, 'upper_ctrl_limit': 32000.0, 'units': 'degrees', 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=0.0, timestamp=1581718543.941563, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}

```

prjemian, 2020-02-14:
Looks like ophyd may be trusting the information from PyEpics.  This shows the limits (in ophyd) being reset after a move.  The information in the `kwargs` is out of date:

```
In [6]: mov m8 0                                                                                                                                                    
new:0 old:-0.1, kwargs={'timestamp': 1581719163.4543722, 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=0, timestamp=1581719163.4543722, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}
new:0.0 old:0, kwargs={'timestamp': 1581719163.454469, 'status': <AlarmStatus.NO_ALARM: 0>, 'severity': <AlarmSeverity.NO_ALARM: 0>, 'precision': 5, 'lower_ctrl_limit': -32000.0, 'upper_ctrl_limit': 32000.0, 'units': 'degrees', 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=0.0, timestamp=1581719163.454469, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}
                                                                                                                                                                    
In [7]: m8.limits                                                                                                                                                   
Out[7]: (-32000.0, 32000.0)

In [8]: mov m8.high_limit_value 1500                                                                                                                                
new high limit:1500.0 old:1500.0, kwargs={'timestamp': 1581719163.682007, 'status': <AlarmStatus.NO_ALARM: 0>, 'severity': <AlarmSeverity.NO_ALARM: 0>, 'precision': 5, 'lower_ctrl_limit': -1e+300, 'upper_ctrl_limit': 1e+300, 'units': 'degrees', 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.HLM', name='m8_high_limit_value', parent='m8', value=1500.0, timestamp=1581719163.682007, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.HLM', limits=False, put_complete=False)}

In [9]: m8.limits                                                                                                                                                   
Out[9]: (-32000.0, 1500.0)

In [10]: mov m8 .1                                                                                                                                                  
new:0.1 old:0.0, kwargs={'timestamp': 1581719201.2541802, 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=0.1, timestamp=1581719201.2541802, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}
new:0.1 old:0.1, kwargs={'timestamp': 1581719201.254226, 'status': <AlarmStatus.NO_ALARM: 0>, 'severity': <AlarmSeverity.NO_ALARM: 0>, 'precision': 5, 'lower_ctrl_limit': -32000.0, 'upper_ctrl_limit': 32000.0, 'units': 'degrees', 'sub_type': 'value', 'obj': EpicsSignal(read_pv='sky:m8.VAL', name='m8_user_setpoint', parent='m8', value=0.1, timestamp=1581719201.254226, tolerance=1e-05, auto_monitor=False, string=False, write_pv='sky:m8.VAL', limits=True, put_complete=False)}
                                                                                                                                                                    
In [11]: m8.limits                                                                                                                                                  
Out[11]: (-32000.0, 32000.0)
```

Limits were just reset.  Based on the kwargs?  Time to look at PyEpics without ophyd.

```
In [13]: epics.__version__                                                                                                                                          
Out[13]: '3.4.1'
```

prjemian, 2020-02-15:
There is a `_metadata_changed()` method that will make all the necessary changes. It needs the PV name and a dictionary of the new metadata.

klauer, 2020-02-15:
For whatever reason, there is no single CA get/monitor request that contains valid timestamps, alarm severity, and limits all in one. You can monitor the limits in EPICS without issue, assuming you use a `DBR_CTRL_*` type. However, ophyd insists that timestamps are more important and uses `DBR_TIME_*` type requests instead. (* see below)

This all means that limits would have to be updated through an external mechanism here (e.g., by monitoring LLM/HLM fields).

(*) For reference,
Control: https://github.com/caproto/caproto/blob/1a7356487fd930cb6dee6c26f1ce1f117cd87c95/caproto/_dbr.py#L606
Time: https://github.com/caproto/caproto/blob/1a7356487fd930cb6dee6c26f1ce1f117cd87c95/caproto/_dbr.py#L425
Graphics: https://github.com/caproto/caproto/blob/1a7356487fd930cb6dee6c26f1ce1f117cd87c95/caproto/_dbr.py#L493

prjemian, 2020-02-15:
So, like this but limit the dictionary (from `self.user_setpoint._read_pv.get_ctrlvars()`) to just the limits?

```

class EpicsMotorLimitsMixin(DeviceMixinBase):
    
    soft_limit_lo = Component(EpicsSignal, ".LLM")
    soft_limit_hi = Component(EpicsSignal, ".HLM")

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

        def cb_limit_changed(value, old_value, **kwargs):
            """
            update EpicsSignal object when a limit CA monitor received
            """
            if (
                self.connected 
                and old_value is not None 
                and value != old_value
            ):
                self.user_setpoint._metadata_changed(
                    self.user_setpoint.pvname,
                    self.user_setpoint._read_pv.get_ctrlvars(),
                    from_monitor=True,
                    update=True,
                    )

        self.soft_limit_lo.subscribe(cb_limit_changed)
        self.soft_limit_hi.subscribe(cb_limit_changed) 
```

klauer, 2020-02-18:
Yes, something like that could work.

prjemian, 2020-03-20:
I will make a branch to combine this [Mixin]( https://github.com/BCDA-APS/apstools/blob/d6298262a3501b48354c6aadb210477834e55b38/apstools/devices.py#L1141-L1215) into the EpicsMotor

klauer, 2020-03-20:
The only issue with this is that it applies only to motors, not other PVs with control limits.

prjemian, 2020-03-20:
Is there a way to know?  The IOC provides a `.RTYP` field that informs if a record instance is a `motor` record (or other).  Need to access this value in the setup for the limits.  Some other way?

prjemian, 2020-03-20:
EPICS motor seems to be the unusual record type that does not HOPR/LOPR.  It uses HLM/LLM.

klauer, 2020-03-20:
I think we can rely on the control limits of the PVs with an additional monitor of type `DBR_GR/CTRL`, as I briefly hinted at in #824. Then it will be agnostic of RTYP.

prjemian, 2020-03-20:
I accede to your skill in this.  Looks much better.

prjemian, 2020-03-20:
If we update the limits in python, how to pass that back to EPICS?  Our clients expect that behavior.

klauer, 2020-03-20:
I don't see any harm in using your proposal in the short-term.

Yeah, I think we'd need separate signals (specifically to HLM/LLM or LOPR/HOPR) where we'd want to be able to update limits from ophyd, as EPICS CA doesn't allow for writing control limits of a PV.

With `EpicsMotor` which map to one `motor` record per instance, having these limit components works out well. For other signals, not so much - we'd have to think about it.

prjemian, 2020-03-20:
I agree that this should only modify EpicsMotor.  I'll continue with the branch, then.

PR #827 (2020-04-04): ENH #802 connect EPICS soft motor limits
Fixes #802 by connecting to EPICS motor record soft limits (`.HLM` and `.LLM`) and keeping ophyd `EpicsMotor` object synchronized with EPICS.

Also supplies two methods for call compatibility with SPEC.
closed 2020-04-04 by prjemian