bluesky/ophyd #807: Decorate `position` property to wait for connection

danielballan, 2020-02-22:
A beamline profiles has:

```py
class ZonePlate(Device):
   ... various EpicsMotor Components ...

zp = ZonePlate(...)

...

blah = zp.position
```

In ophyd 1.3.x, this works. In 1.4.0rc4, it fails on a connection error. Like `read()`, `position` should block for connection (up to the connection timeout) rather than promptly raising an error without giving some time for the connection to complete.

danielballan, 2020-03-04:
It looks like this is decorated with `raise_if_disconnected`, which immediately raises without trying to connect. I'm a little confused about the right resolution here.

klauer, 2020-03-04:
Properties that block on access are awful, in my opinion.

I think if you intend to use a device shortly after instantiation, it's not so bad to require a `.wait_for_connection()` call - or a check if it's connected prior to accessing. get/put/read/set are at least explicit in that there are timeouts involved.

danielballan, 2020-03-04:
That is a solid point. I fully agree. Closing as “won’t fix”, where the recommended path is as @klauer described above. This is indeed exactly what @mrakitin did when he encountered this in the wild.

It sounds like 0.4.0 is exposing a race condition that has always been lurking but it now easier to hit. Best to address it properly by explicitly waiting.

Thanks, @klauer.

closed 2020-03-04 by danielballan

mrakitin, 2020-03-04:
It was @tacaswell who showed it to me first 😃